realsenseai/librealsense #9203: Not able to stream the depth or color frame in XGA or VGA Resolutions

sowmiyadharmalingam86, 2021-06-11:
| | |
|---|---|
|**librealsense**|2.47.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1062528|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|L500|
|**Asic Serial Number**|0003ac531045|
|**Firmware Update Id**|0003ac531045|

Please provide a description of the problem

I am able to stream the data  with 320*240 (depth) & 640*480(RGB). The XGA(640*480) or VGA(1024*768)  resolutions  not able to stream either by the real sense viewer or by SDK2.0

MartyG-RealSense, 2021-06-27:
Hi @sowmiyadharmalingam86  The data sheet document for the L515 camera model confirms the depth and RGB resolution settings that are supported.

https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet

**Depth**
[image]
**RGB**
[image]

So you should be able to stream in VGA (640x480) and XGA (1024x768) for depth, but not access those resolutions on RGB as they are not on the list of supported RGB resolutions.

[image]

MartyG-RealSense, 2021-07-09:
From L515 firmware version **1.5.8.1** onwards, 640x360 and 640x480 RGB resolution support is added.

[image]

closed 2021-07-20 by MartyG-RealSense