realsenseai/librealsense #9208: Read color_frame from .bag file

sn1c3, 2021-06-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (v2.47.0) | 
| Operating System & Version |   {Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-74-generic                                | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/AR }                          | 

### Issue Description
This should be a pretty easy question, I just did not find an answer online to it. So I basically want to edit the color and depth frame from a bag file. Therefore, I have modified the "read_bag_example.py" script in a way that I can view both frames.

#### Bag file
I start the realsense viewer with `realsense-viewer`. After that I turn on the stereo module and the rgb camera and click record.

####  Python Code
 The modified code looks like this:

```

#####################################################
##               Read bag from file                ##
#####################################################

# First import library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2
# Import argparse for command-line options
import argparse
# Import os.path for file path manipulation
import os.path

try:
    # Create pipeline
    pipeline = rs.pipeline()

    # Create a config object
    config = rs.config()

    # Tell config that we will use a recorded device from file to be used by the pipeline through playback.
    # 20210607_122415
    path = "<path>/20210612_130610.bag"
    rs.config.enable_device_from_file(config, path)

    # Configure the pipeline to stream the depth stream
    # Change this parameters according to the recorded bag file resolution
    config.enable_stream(rs.stream.depth, rs.format.z16, 30)

    # Start streaming from file
    pipeline.start(config)

    # Create opencv window to render image in
    cv2.namedWindow("Depth Stream", cv2.WINDOW_AUTOSIZE)
    
    # Create colorizer object
    colorizer = rs.colorizer()

    # Streaming loop
    while True:
        # Get frameset of depth
        frames = pipeline.wait_for_frames()

        # Get depth frame
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        
        if not depth_frame:
            continue

        # Colorize depth frame to jet colormap
        depth_color_frame = colorizer.colorize(depth_frame)

        # Convert depth_frame to numpy array to render image in opencv
        depth_color_image = np.asanyarray(depth_color_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        images = np.hstack((color_image, depth_color_image))

        # Render image in opencv window
        cv2.imshow("Depth Stream", images)

        # if pressed escape exit program
        if cv2.waitKey(1) == 27:
            cv2.destroyAllWindows()
            break

finally:
    pipeline.stop()

```

#### Problem

Somehow I can not access the color frame. Error log:
```

(env_jvirt) user:~/joint_visualization$ python scripts/read_bag_example.py 
Traceback (most recent call last):
  File "scripts/read_bag_example.py", line 61, in <module>
    color_image = np.asanyarray(color_frame.get_data())
RuntimeError: null pointer passed for argument "frame_ref"

```

MartyG-RealSense, 2021-06-12:
Hi @sn1c3  The link below has information about adapting **read_bag_example.py**, including a link to a script that a RealSense user created to use color in this program instead of depth by adding a config_enablestream definition for color.  This will hopefully be a helpful reference for creating an adaptation that can access both depth and color.

https://github.com/IntelRealSense/realsense-ros/issues/1836#issuecomment-833826374

sn1c3, 2021-06-13:
I solved it by just commenting out the config.enable_stream() lines.
```

#####################################################
##               Read bag from file                ##
#####################################################

# First import library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2
# Import argparse for command-line options
import argparse
# Import os.path for file path manipulation
import os.path

# Create pipeline
pipeline = rs.pipeline()
config = rs.config()

# Tell config that we will use a recorded device from file to be used by the pipeline through playback.
path = "20210607_122415.bag"
rs.config.enable_device_from_file(config, path)

# Configure the pipeline to stream the depth stream
# Change this parameters according to the recorded bag file resolution
# config.enable_stream(rs.stream.depth, rs.format.z16, 30)

# Start streaming from file
pipeline.start(config)

# Create opencv window to render image in
cv2.namedWindow("Depth Stream", cv2.WINDOW_AUTOSIZE)

# Create colorizer object
colorizer = rs.colorizer()
try:
    # Streaming loop
    while True:
        # Get frameset of depth
        frames = pipeline.wait_for_frames()

        # Get depth frame
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        
        if not depth_frame:
            continue

        # Colorize depth frame to jet colormap
        depth_color_frame = colorizer.colorize(depth_frame)

        # Convert depth_frame to numpy array to render image in opencv
        depth_color_image = np.asanyarray(depth_color_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        images = np.hstack((color_image, depth_color_image))

        # Render image in opencv window
        cv2.imshow("Depth Stream", images)

        # if pressed escape exit program
        if cv2.waitKey(1) == 27:
            cv2.destroyAllWindows()
            break
        
finally:

    # Stop streaming
    pipeline.stop()

```

closed 2021-06-13 by sn1c3

MartyG-RealSense, 2021-06-13:
Great to hear that you developed a solution @sn1c3 - thanks very much for sharing it with the RealSense community!