realsenseai/librealsense #9211: D455 not turn on

Framemov, 2021-06-12:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

Hello,
Today was the first day that I started working with a d455. Everything was working properly. After testing it, I turned off the computer with the camera connected. After two hours I turned on the computer again but the camera is no longer found on my Windows 10 computer and neither on any of the other computers in the office, all of them with Windows 10.

When I connect the USB nothing happens. Windows does not recognize any connection from the device. I have tried different ports, different cables, different computers but nothing happens. It is as if the camera is not able to turn on.

Could you please tell me if there is something I can do or is it just a camera hardware error?

Thank you
Greetings

MartyG-RealSense, 2021-06-12:
Hi @Framemov   Can you confirm whether you have opened the Windows **Device Manager** interface and looked for the camera's two RGB and Depth drivers under the **Cameras** category after having plugged the camera into the USB port, please?

[image]

If you are able to access the Device Manager and the two drivers are not present in the **Cameras** category, please next try checking the camera **Privacy** settings of Windows to confirm that camera positions are set to **On** using the instructions in the link below.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

If the camera is not detected by any of the cameras in the office then it is probably not a privacy settings issue, but it is quick to check and worth doing just to eliminate the possibility.

Framemov, 2021-06-12:
Hi MartyG,
Thanks for your quick response.
There is nothing in device manager about cameras. 
I have plugged the device into a usb 3.2 port and a usb c cable, where it previously worked perfectly and it works perfectly with t265, but when I connect the 455, alone, nothing happens. 
There is no sound of external device connection in windows, nor any change in the usb bus information.
[image]

MartyG-RealSense, 2021-06-12:
And it is not detected if you unplug the micro end of the USB cable from the side of the camera, turn the cable upside down and re-insert into the side of the camera? (USB C cables are two-way insertion at the micro end)

Framemov, 2021-06-12:
nothing change :(

MartyG-RealSense, 2021-06-12:
Could you clarify the statement about T265 please - the D455 works if both it and the T265 are plugged in, but the D455 does not work if plugged in on its own?

Framemov, 2021-06-12:
sorry, my previous message was unclear. d455 never works, in neither case

MartyG-RealSense, 2021-06-12:
Have you tried sequential unplug-replugs of the cable?  Unplug, wait a couple of seconds and re-plug, doing this about 10 times to see whether one of these multiple insertions results in the insertion sound effect.  If you have a spare USB port, try alternating the insertion - plug into one, remove and plug into the other, then back to the first port.

Framemov, 2021-06-12:
Tried on different computers but still nothing happens. It is as if it does not receive electricity but the cable and ports work with other devices.

MartyG-RealSense, 2021-06-12:
At this point I will seek advice from Intel on Monday - please leave a comment for me here on Monday as a reminder.  It would be best to put the camera aside until then, as I do not believe that there is anything else that you could try during the weekend.

Framemov, 2021-06-12:
Thank you very much for your time Marty. I'll send you a reminder on Monday. happy weekend

MartyG-RealSense, 2021-06-14:
Hi @Framemov  I have communicated with Intel regarding your D455 detection problem.  I will update you when I have information.  Thanks for your patience!

MartyG-RealSense, 2021-06-14:
Could you send me your email address so that I can pass it on to Intel, please?  My email address is below:

martyx.grover@intel.com

MartyG-RealSense, 2021-06-17:
Hi @Framemov   An Intel RealSense representative emailed you on Monday 14 June.  Did you receive the message please?  If you did not, can you check the spam / junk folder of your email box to see whether it is in there.  Thanks very much.

Framemov, 2021-06-17:
Hi Marty,
Yes, I have received the email and I am already in the process of return.
I think they contacted me in several ways perhaps beacuse i have two mails.
I am waiting for the prepaid label  to send the camera, but for the moment it has not arrived.
Thanks for your time and concern in this matter
J,

Jorge Escobar
Director Creativo
+34 606 358 701
http://www.framemov.com/

MartyG-RealSense, 2021-06-17:
The representative who contacted you is waiting to receive your shipping address so that they can request a pre-paid shipping label for you.

MartyG-RealSense, 2021-08-23:
Hi @Framemov   Can you confirm if this returns case is fully resolved and able to be closed now, please?  Thanks!

Framemov, 2021-08-23:
Hi Marthy,
Yes, we have finally solved the problem and they have returned my money because I bought other cameras to be able to arrive on time with the delivery of the project. Everything is solved and I am very happy with the attention of the intel supprt.
Thank you
Cheers

Jorge Escobar
Director Creativo
+34 606 358 701
http://www.framemov.com/

manic1841, 2025-02-07:
Hi @Framemov, could you share your experience on how you solved the problem? Thanks. I have the same issue as you.