realsenseai/librealsense #9224: D435 linux update FW with C API error

csp23, 2021-06-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                      | 
| Firmware Version                   | 05.12.07.100                                           | 
| Operating System & Version | Linux                                                       | 
| Kernel Version (Linux Only)    | 4.9.37                                                      | 
| Platform                                 |  raspberry                                               |
| SDK Version                            |  2.42.0                                                    | 
| Language                            |  C                                                              | 
| Segment			|  Robot                                                            | 

### Issue Description
<Describe your issue / question / feature request / etc..>

#### fail with API  rs2_update_firmware()  from rs_device.h

function using( reference example/C/rs-depth.c ) :
rs2_creat_context()..........create_context
-->rs2_query_devices().........get device_list
---->rs2_create_device()..........get device
------>rs2_update_firmware()....update with device and FW img

* error message:
**rs_error was raised when calling rs2_update_firmware(device:0xed18a0, fw_image:0x7f72bff010):
    Object does not support "librealsense::update_device_interface" interface!** 
* my opinion:
the object  return from function _rs2_create_device()_ dose support update FW
* ps:
i could successfully update with tools/rs-fw-update  but still hope using API

MartyG-RealSense, 2021-06-15:
Hi @csp23  I do not have a C-specific example of updating firmware through scripting, though some general advice about structuring an updater mechanism with **update_device()** is provided in the link below.

https://github.com/IntelRealSense/librealsense/issues/4416#issuecomment-511143865

An example of how a firmware updater script may be structured in Python is here:

https://github.com/IntelRealSense/librealsense/issues/4416#issuecomment-511481279

It is also worth bearing in mind that the firmware file that is being used should be a **digitally signed** one - a backup firmware file will not be sufficient.

https://github.com/IntelRealSense/librealsense/issues/7093#issuecomment-672186337

Although checking and updating firmware through scripting is supported, updating with the **rs-fw-update** tool or the RealSense Viewer are the recommended methods.

csp23, 2021-06-15:
Thanks a lot

i fix it with cast device first

* it worked with API using:

rs2_creat_context()..........create_context
-->rs2_query_devices().........get device_list
---->rs2_create_device()..........get device
**------->rs2_enter_update_state()......change device state and reconnect realsense on usb**
--------->rs2_delete_device();
----------->rs2_delete_device_list();
------------->rs2_delete_context();
==============get reconnected device=========== 
rs2_creat_context()..........create_context
-->rs2_query_devices().........get device_list
---->rs2_create_device()..........get device
------>rs2_update_firmware()....update with device and FW img

done

MartyG-RealSense, 2021-06-15:
Great to hear that you were successful @csp23 - thanks very much for sharing your C method with the RealSense community!

closed 2021-06-21 by MartyG-RealSense