realsenseai/librealsense #9229: D435i tests based on DQT

camiloupegui, 2021-06-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.13.50 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                     | 
| Platform                                 | PC  |
| SDK Version                            |  2.47.0   | 
| Language                            |  python/C++                        | 
| Segment			|  others                         | 

### Issue Description

I am currently trying to set a protocol to test some realsense D435i cameras. I want to reuse the tools used in the DQT adapting them to my program and protocol, and of course following the metrics from the DQT paper. However, I have some questions regarding the required distances for setting the test environment.
1.	If we test the camera with a target at 50cm and it meets all the performance criteria, can we ensure that the camera is working properly? Or would it be better to set the target at different distances and do some more tests to completely ensure the full functioning of the camera?
 
2.	Like the first one, if we test the camera with a ROI of 20% or 40% and it meets all the performance criteria, can we say that both the tested area and the area that doesn’t enter the ROI are functioning well? Or it is necessary to test the camera with a ROI of 80% to ensure that all the pixels are receiving the right information?

I need to meet some requirements for this protocol, so it would be perfect to find the balance between quality of the test and minimum space required to set the test environment. By the way, the final application is a mobile robot.

MartyG-RealSense, 2021-06-15:
Hi @camiloupegui   Intel have published a **Depth Testing Methodology** paper in PDF format (updated in January 2021) that describes a depth testing procedure for RealSense 400 Series cameras.

https://dev.intelrealsense.com/docs/camera-depth-testing-methodology

Page 13 onwards of the document describes testing in terms of the **Depth Quality Tool**.

In regard to question 1, page 14 (section **2.4.1  Target Distance**) of the document advises to place the camera betwen 50 cm and 2 meters away from the target surface (longer-distance placement is possible but potentially more problematic).  The Depth Quality Tool can provide feedback, such as the current amount of **RMS error** (error that increases linearly as distance from the camera increases).  

Page 9, **Section 1.3.3  RMS Error (Spatial Noise)** provides more information about RMS Error.

In regard to question 2. page 15 of the paper advises selecting a ROI size that is appropriate for the **target size and distance**.  the paper states that "the same target that works at 1 meter with 80% ROI might only cover 40% at 2 meters".

camiloupegui, 2021-06-15:
Thank you for the answer. I have already read most of that document, however, my question is more directed towards the influence that choosing one ROI or another have in the resulting values of the test. For example, can we validate a camera when it meets all the performance criteria, no matter if the test was with ROI of 80% or 40%? since in the second one we are evaluating less pixels and that somehow will give you less certainty. 

in short we need to ensure that at minimum values of the test environment like 50cm distance and 40% ROI we are still able to ensure that the camera is completely proved, even if we are not evaluating the whole measurement region.

MartyG-RealSense, 2021-06-15:
Are the environment and lighting conditions going to be consistent for every test, please?  Changes in conditions such as lighting and air temperature could introduce variables into the test that affect accuracy.  So even if the performance criteria were precisely replicated for each test, changes in environment could alter the outcome.

camiloupegui, 2021-06-15:
Well, the idea is to set the environment to ensure consistent conditions of lighting and temperature during all the tests, although it is difficult to guarantee that. But for setting that we need to know if those minimum values of distance and ROI will give us the expected results, in other words if we could trust the values that we are obtaining to indicate if the camera is working properly or not.

MartyG-RealSense, 2021-06-15:
Setting a ROI and analyzing it may be more complexity than the problem requires.  If you set up a test where the camera is stationary and you know the real-world distance between the camera and the object then checking a text readout of distance would provide an indicator of how close the depth measurement is to the known true distance from camera to object (bearing in mind that factors such as RMS error mean that you are unlikely to achieve precisely the known real-world measured distance to the object).

Bear in mind that on the cased camera models (D415, D435/i, D455), depth is measured from the front glass instead of the camera lens, so a minus-value should be added to the measurement to find the **ground truth** depth measurement.

In the Depth Quality Tool (and also the RealSense Viewer), you can obtain the XYZ coordinates of a position that the mouse cursor is at as a text overlay.

[image]

A simpler way to obtain the distance is as a text-only readout in the **rs-hello-realsense** example program.

[image]

camiloupegui, 2021-06-16:
The problem is that we also want to test the other metrics like the fill-rate or RMS error and not just the distance between the camera and the target. For instance, part of the idea is to do both the passive and active tests to see if the projector is also working properly, in this case the Fill-rate is even more important and for that we need to set the ROI to read just the area of the interchangeable target ( white/textured). But again the question here is if we can guarantee that the entire measurement region of the sensor is working properly just by evaluating for example 50% (ROI) of this region.

MartyG-RealSense, 2021-06-16:
The default settings of the SDK are already purposely configured by Intel to provide very good results and you should not need to define a custom ROI unless there is a specific reason to do so (such as focusing on a particular area of the image or dealing with outdoor sunlight).  So if you wish to benchmark performance against the ideal expected behavior of the camera then leaving the settings on their defaults may be the best way to do so.

MartyG-RealSense, 2021-06-22:
Hi @camiloupegui  Do you require further assistance with this case, please?  Thanks!

camiloupegui, 2021-06-23:
Hi, I don't need further assistance, thank you!

MartyG-RealSense, 2021-06-23:
You are very welcome @camiloupegui - thanks very much for the update!  I will close this case now as you are satisifed with the outcome.

closed 2021-06-23 by MartyG-RealSense