realsenseai/librealsense #9232: wrong box dimension

yuanyuanhan327, 2021-06-15:
Hello, after updating firmware, the box_dimensioner_multicam_demo.py is wrong. Please see the picture. Do you know the problem?

[image]

yuanyuanhan327, 2021-06-16:
Thanks MartyG. 

1. I am using D455. 
2. Yes, it is empty. After finishing calibrating, I put the objects on the chessboard.
3. Yes, I used 6*9 chessboard
4. It seemed to improve a little if I move the camera far away (>=50cm)

i checked and I don't know why the depth is 0 when the distance between cameras and objects is about 20,30cm.

MartyG-RealSense, 2021-06-16:
Thanks very much for the information.

Your results seem to match the case that I linked to earlier, where results improve at around 0.5 meters distance from the board but fail at 1 meter.

https://support.intelrealsense.com/hc/en-us/community/posts/360044001733/comments/1500001064501

The key factor is likely to be the camera model that you are using though.  The D455 has a minimum depth sensing distance of 0.4 meters (40 cm), whilst the D435 and D435i have a minimum depth distance of 0.1 m (10 cm) that enable the camera to get closer to objects before the depth detail starts breaking up below minimum distance.

The minimum depth of a camera can be reduced by reducing the depth resolution (e.g from 1280x720 to 848x480, which is actually the optimal depth accuracy resolution for D455)

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py#L30

Minimum distance can alternatively be reduced by increasing the value of a setting called **Disparity Shift**.  A Python example for setting Disparity Shift is in the link below. 

https://github.com/IntelRealSense/librealsense/issues/2015

educatornum, 2021-07-06:
@yuanyuanhan327 Hello Sir,

educatornum, 2021-07-06:
@yuanyuanhan327  Hello Sir,
Did you use two Intel® RealSense™ Depth Cameras D435 cameras?

educatornum, 2021-07-07:
@MartyG-RealSense @yuanyuanhan327 Hello, 
I am new to RealSense must I need to buy two Intel® RealSense ™ Depth D435 cameras to will box dimensions?

MartyG-RealSense, 2021-07-07:
Hi @educatornum  Whilst the **box_dimensioner_multicam** program was designed for use with more than one camera, it does work with a single camera.

educatornum, 2021-07-15:
@MartyG-RealSense Hello Sir, 
Thank you for your advice. It is worked.

And I have one important question. 
I want to develop a web application that uses this box dimension python script and OpenCV.  I tried it but it does not work on the web. 

> I used ReactJS, OpenCV2, and Realsense D345.

Is it possible?  Could you give me some advice?

MartyG-RealSense, 2021-07-15:
If you need to access a Python application through the web, a **Flask server** may meet your needs.  Resources about using RealSense and Python with Flask, and also using RealSense in a browser with HTML, can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/9291

closed 2021-09-17 by MartyG-RealSense