realsenseai/librealsense #9233: PyPi arm64 Python builds/wheels

DaAwesomeP, 2021-06-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A | 
| Firmware Version                   | N/A | 
| Operating System & Version |  Linux  | 
| Kernel Version (Linux Only)    |  5.3.18-57-preempt                                         | 
| Platform                                 | Raspberry Pi |
| SDK Version                            |  N/A                        | 
| Language                            |  Python3                          | 
| Segment			|  Robot/VR/AR/others                          | 

### Issue Description

Add Python wheel builds on PyPi for Linux arm64/aarch64 (Raspberry Pi, etc.). I am happy to submit a pull if I can be pointed in the right direction for how this repo's automatic build system for wheels works. It can definitely be cross-compiled from amd64.

MartyG-RealSense, 2021-06-16:
Hi @DaAwesomeP   The installation section of the Python wrapper documentation states that PyP distribution files are created by running the wrapper's **setup.py** file with the command **python setup.py bdist_wheel**

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#installation

[image]

DaAwesomeP, 2021-06-17:
OK, I'm working on it but I am not able to run `bdist_wheel` even on amd64. I compile with Python enabled and run `make install` (and the library works when I use it in Python files), but when I try to make the wheel I get:
```bash
# python3 setup.py bdist_wheel
Traceback (most recent call last):
  File "setup.py", line 5, in <module>
    from pyrealsense2._version import __version__
  File "/home/pi/librealsense/wrappers/python/pyrealsense2/__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'
```

MartyG-RealSense, 2021-06-18:
@DaAwesomeP  I do not have knowledge of compiling Python wrapper wheels beyond the use of **bdist_wheel** so I will tag @Nir-Az to seek input about this case.

Nir-Az, 2021-06-21:
Hi @DaAwesomeP ,
According to the error the script doesn't find pyrealsense binaries.
You need to copy the PYD/SO (Windows/Linux) to this path "wrappers/python/pyrealsense2"
and run this command:

```
cd wrappers/python
python setup.py bdist_wheel

```

Please try and update..

Good luck

MartyG-RealSense, 2021-06-21:
Thanks very much @Nir-Az for your kind assistance with this case!

MartyG-RealSense, 2021-07-04:
Hi @DaAwesomeP  Do you require further assistance with this case, please?

DaAwesomeP, 2021-07-04:
Hello! I have made a lot of progress, but not quite done yet. I am working on it a bit each week. The main thing is that the full compile process must be initiate-able from `setup.py`. I have it building, but the packages don't work yet; they don't have the right binaries in them.

Here is an example run:
https://github.com/DaAwesomeP/librealsense/actions/runs/965029724

MartyG-RealSense, 2021-07-04:
Thanks very much @DaAwesomeP for the update.  I will keep this case open for a further time period whilst you are working on the project.

@Nir-Az  Do you have advice that you can offer @DaAwesomeP please about their issue with the packages not having the right binaries in them, please?

Nir-Az, 2021-07-04:
> Thanks very much @DaAwesomeP for the update. I will keep this case open for a further time period whilst you are working on the project.
> 
> @Nir-Az Do you have advice that you can offer @DaAwesomeP please about their issue with the packages not having the right binaries in them, please?

This looks like CI issues that are far beyond my scope.
We got an example on how to build the python packages on our documentation.

Let me know if I can help with something else.

MartyG-RealSense, 2021-07-10:
Hi @DaAwesomeP  Do you require further assistance with this case, please?  Thanks!

DaAwesomeP, 2021-07-14:
Hi Marty, this will take some time to complete.

MartyG-RealSense, 2021-07-14:
Okay, no problem at all.  If it is going to be weeks rather than days, do you want to close this case and re-open it later or make a new issue?  Thanks!

DaAwesomeP, 2021-07-14:
Sure

closed 2021-07-14 by DaAwesomeP