realsenseai/librealsense #9241: Min Distance for D455

yuanyuanhan327, 2021-06-16:
Hello, when I used D455, both the depth resolution and the color resolution are set to 1280x720. But the depth for objects (distance < 50cm) is 0. How can I solve it? Thanks.

agrunnet, 2021-06-16:
For best depth we recommend using 848x480 for that camera. That will reduce the MinZ to close to 34cm.

MartyG-RealSense, 2021-06-16:
@agrunnet  Thanks very much for helping @yuanyuanhan327   :)

@yuanyuanhan327  The advice of @agrunnet is correct.  Reducing resolution reduces the camera's minimum distance (MinZ), and 848x480 is also the optimal depth accuracy resolution for D455.

agrunnet, 2021-06-16:
And you can also use the shifting of disparity under advanced functions to get even closer. But please read our white paper on all the recommendations. It is on the website.

MartyG-RealSense, 2021-06-22:
Hi @yuanyuanhan327  Do you require further assistance with this case , please?  Thanks!

closed 2021-06-28 by MartyG-RealSense