realsenseai/librealsense #9243: Unable to change the firmware version  in L515

jrtk, 2021-06-17:
we are unable to change/downgrade to the stable major firmware version 1.5.0.0 from the minor version 1.5.4.1 in  L515 Camera. I tried with both the RealSense Viewer and also with the tools from rs-fw-update.  I end up with following error. 

Started Firmware Update process
Recovery device connected, starting update
ERROR: Invalid Value in rs2_update_firmware_cpp(device:0x7ff6f0acd5f0, fw_image:0x555746da70a0):
Unsupported size provided { 2098076 }, expecting { 1573660 }

Host Machine:

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.5 LTS"

Could you please help?

RealSenseSupport, 2021-06-18:
Hi @jrtk 
from the logs, it shows the firmware you used has wrong file size (2098076)
you can download the firmware from below link for L515 and which size should be 1537k
https://dev.intelrealsense.com/docs/firmware-releases-l500

RealSenseCustomerSupport, 2021-06-30:
----------------------------------------------

Is this problem resolved?

closed 2021-07-14 by RealSenseCustomerSupport