realsenseai/librealsense #9244: Inconsistent exposure when set manually using pyrealsense2

rkcroc, 2021-06-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i              | 
| Firmware Version                   | 05.12.13.50     | 
| Operating System & Version |   Win 10          | 
| Kernel Version (Linux Only)    |  N/A               | 
| Platform                                 |  PC                  |
| SDK Version                            |  2.45.0.3212   | 
| Language                            |  python              | 
| Segment			|  Manufacturing       | 

### Issue Description
I am using a python script to test image quality with various exposure and gain settings. When exposure is set manually, I sometimes get exposure times that are not what I have set. My script outputs a warning in those instances, e.g.:
```
Image 24
  option_enable_auto_exposure = 0.0
  option_exposure = 39.0
  option_gain = 16.0
  Warning: requested and actual option values do not match:
      exposure_get=39.0 (50.0), gain_get=16.0 (16.0)
```
In this case, I am doing `sensor.set_option(rs.option.exposure, 1000)`, but `sensor.get_option(rs.option.exposure)` is returning 1250. This behavior also occurs with exposure set to 100, 500, and 1000, resulting in returned exposures of 78, 625, and 1250 respectively. Exposure set to 5000 or 10000 does not show this problem; the same exposure time is returned.

I have attached my script, and the output I am seeing.
[exposureIssue.zip](https://github.com/IntelRealSense/librealsense/files/6672541/exposureIssue.zip)

MartyG-RealSense, 2021-06-18:
Hi @rkcroc   It looks as though exposure and gain is being changed on every image in your test script, and it performs this process a maximum of 41 times.  Is this correct please?

If this is how the script works then I would recommend having a sleep period between every image.  This is because when an rs.option setting such as exposure is changed, it can take several frames for the change to complete.

rkcroc, 2021-06-18:
Yes, the 41 images with different gain & exposure is as intended. We are evaluating the camera's autoexposure versus manual settings.
I changed the script to `sleep(2)` after starting the pipeline; that didn't change the reported exposure times.

MartyG-RealSense, 2021-06-18:
I see that you are reading the exposure and gain values from metadata, which is the appropriate method to use.  You have also defined the RGB sensor properly using **first_color_sensor** in order to ensure that it is the RGB sensor and not the depth sensor that is being accessed.

I note that your color auto-exposure enable instruction seems to be within the streaming loop.

```
color_sensor.set_option(rs.option.enable_auto_exposure, 1)
color_sensor.set_option(rs.option.auto_exposure_priority, 1)
else:  # manual exposure and gain
color_sensor.set_option(rs.option.enable_auto_exposure, 0)
```

If it is within the loop, I can see the possibility of color auto-exposure being re-enabled on each loop-through.  This could cause the exposure value to jump away from the manual value.  When auto-exposure is enabled, it also takes several frames for the exposure to settle down to the correct value.  

This is why a common practice when using auto-exposure is to skip the first several frames of the stream after the pipeline is started to avoid frames with bad exposure values.  If enabling auto-exposure on each run-through is an intended behavior of this script, the time taken for exposure to settle after enablement may make this method impractical for this particular application.

rkcroc, 2021-06-18:
@MartyG-RealSense 
During each loop the script is either turning on auto exposure, or turning it off and setting gain and exposure manually. How exactly would it be re-enabled? Also, note that I am starting and stopping the pipeline after gathering frames at each gain & exposure setting. Is this the correct way to go about changing the settings?

BTW, this script is for testing purposes, not for production, so spending time to wait for exposure to settle is not an issue.

MartyG-RealSense, 2021-06-19:
My interpretation of the script is that color auto-exposure is enabled only IF the exposure value = '0', ELSE if exposure does not equal '0' then manual exposure and gain values are set.  The manual values that are set are the ones currently stored in the 'exposure' and 'gain' variables.  BUT the setting of those manual values can only be triggered if exposure is detected to = '0', otherwise the code within the IF statement could not be triggered.  

You are enabling and disabling color auto-exposure because you want to test with both automatic and manual exposure to see the difference between different modes and exposure / gain values.

If exposure = 0 then it looks like color auto-exposure is enabled.  But on the next time it is checked, if color auto exposure is enabled then the ELSE statement toggles auto-exposure off and uses manual exposure / gain values.  On the next check after that, color auto-exposure would be toggled on again (i.e if the 'exposure = '0' condition is met, auto-exposure is toggled on, off, on, off as the loop through the 41 cycles progresses).

[image]

Whenever exposure = '0', a **sleep** period of '5' activates.  So that sleep period should occur both for auto-exposure and manual values, since activation of it is only dependent on whether exposure = '0', not whether auto-exposure is enabled or not.

After each of the 41 cycles in the loop, once the values are printed then the pipeline is stopped, new settings are applied and then the pipeline is re-started after that.

Does that sound correct, please?  Please correct me on any errors in interpretation.

It is worth bearing in mind if it is not taken account of already that whilst auto-exposure is enabled, the exposure value is not being updated.  This is why it is necessary to read the real exposure value from metadata.  When auto-exposure is disabled then the exposure returns to the **last known** manual values.

closed 2021-07-01 by MartyG-RealSense