realsenseai/librealsense #9245: 3D face stripes

nevynwong, 2021-06-18:
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.12.11.00 | 

The effect of 3D facial reconstruction captured by me is as follows. How to improve it?
[image]

MartyG-RealSense, 2021-06-18:
Hi @nevynwong  The situation with this subject remains the same as the previous case https://github.com/IntelRealSense/librealsense/issues/8558 in which we extensively discussed it.

In that previous case you did not describe the process that you used to create the face image above.  Could you tell me the steps that you take to create it please so that I can research whether there is a way to improve that process, please?

nevynwong, 2021-06-18:
I still haven't improved on this problem, which is the result of a reconstruction of the point cloud that I captured with the RealSense Viewer.I solved the problem last year, but it still didn't work out these two times, and the stripes would still appear.
[image]

nevynwong, 2021-06-18:
Whether or not this is related to light, you can change this phenomenon by supplementing light.

agrunnet, 2021-06-18:
Looks like some quantization. Please try reducing the depth units from 0.001 to 0.0001 (which is 100um steps) and try to set the disparity modulation a-factor to 0.1 (default is 0). Also be sure to use latest FW

nevynwong, 2021-06-18:
I have modified these, but the effect is still not good. Is the stripe related to the lack of light?  I also set the following parameters: the depth unit is 100, Disparity Shift is 70, and the a-factor is 0.08

agrunnet, 2021-06-18:
Are you sure the quantization is not happening somewhere else in your code then?

nevynwong, 2021-06-18:
I've set these parameters, but I don't know what the problem is.------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年6月18日(星期五) 晚上11:21
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [IntelRealSense/librealsense] 3D face stripes (#9245)

agrunnet, 2021-06-18:
Please provide more information. 
What resolution are you using for the depth camera? What range are you measuring the face at? What reconstruction software are you using?

nevynwong, 2021-06-19:
The resolution of the depth camera is 848*480, and the measured distance between the face and the lens is 40cm-60cm. The reconstruction software I used was Matlab Code and MeshLab, and the results were the same.

MartyG-RealSense, 2021-06-19:
I ran extensive further tests but have not been able to create a better solid model than the images in this discussion unfortunately.

agrunnet, 2021-06-26:
[image]
@nevynwong you definitely need to set depth-units to 100. See attached image. It will make the quantization (aka "stripes") disappear. All depth will then be reported in steps of 100um and not steps of 1mm. I tested here and it help.

MartyG-RealSense, 2021-07-08:
Hi @nevynwong  Did the advice provided above about depth units improve the results of your face image please?

nevynwong, 2021-07-13:
I set it this way, but it still doesn't work with the 42 SDK, but it works well with the 24 SDK.

MartyG-RealSense, 2021-07-19:
Hi @agrunnet  Do you have any further suggestions for @nevynwong about reducing face lines if setting depth units to 100 hasn't worked, please?  Thanks!

agrunnet, 2021-07-19:
Not really. There are only there things that would lead to the quantization lines. 
1. the depth units not being small enough 
2. The A factor (disparity modulation) being bad. - try adjusting it from 0 to 0.2. 
3. some quantization or integer math being done somewhere in the users apps or in the visualization.  
Those are the only three to focus on.

MartyG-RealSense, 2021-07-25:
Hi @nevynwong  Could you provide an update about this case please?  Thanks!

MartyG-RealSense, 2021-08-05:
Hi @nevynwong  This case will be closed after 7 days from the time of writing this if we do not hear from you before then.  Thanks very much.

closed 2021-08-16 by MartyG-RealSense