realsenseai/librealsense #9247: face_recognition + librealsense resulting in exit without errors

DieProgrammIDE, 2021-06-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | Irrelevant | 
| Operating System & Version |  L4T Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  Irrelevant                                         | 
| Platform                                 | Nvidia Jetson Xavier NX, Jetson Nano  |
| SDK Version                            |  2.44.0 }                          | 
| Language                            |  Python 3                          | 
| Segment			|  Robot/AI                          | 

### Issue Description
Hello dear Intel Realsense Team, I am building a Robot following a Person, avoiding Obstacles.

For Finding and Tracking the Person, Face-Recognition is required.
For that i use the Python Library [face_recogntion](https://github.com/ageitgey/face_recognition). The only problem is, that when both librealsense (2.44.0) and face_recognition are loaded, certain functions like `align.process(frames)` don't work and cause the Code to just exit without any errors.

For Performance Reasons I am also using Multiprocessing, I have modified the Python-Example `align-depth2color.py` to Import the Library and use Multiprocessing.

Because I didn't find the means to activate debugging for librealsense, I couldn't conduct more debugging.
Looking forward to your help!

```
## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#####################################################
##              Align Depth to Color               ##
#####################################################

# First import the library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2

import face_recognition

from multiprocessing import Queue
from multiprocessing import Process

def d435realsense():
    context = rs.context()
    pipe = rs.pipeline(context)
    config = rs.config()
    config.enable_device('943222073562')
    # Get device product line for setting a supporting resolution
    pipeline_wrapper = rs.pipeline_wrapper(pipe)
    pipeline_profile = config.resolve(pipeline_wrapper)
    device = pipeline_profile.get_device()
    device_product_line = str(device.get_info(rs.camera_info.product_line))

    found_rgb = False
    for s in device.sensors:
        if s.get_info(rs.camera_info.name) == 'RGB Camera':
            found_rgb = True
            break
    if not found_rgb:
        print("The demo requires Depth camera with Color sensor")
        exit(0)

    
    #config.enable_record_to_file('save.bag')

    width = 848
    height = 480
    framerate = 60

    config.enable_stream(rs.stream.depth, width, height, rs.format.z16, framerate)
    config.enable_stream(rs.stream.color, width, height, rs.format.bgr8, framerate)
    profile = pipe.start(config)

    sensor_dep = profile.get_device().first_depth_sensor()
    sensor_dep.set_option(rs.option.enable_auto_exposure, 1)
    #spatial = rs.spatial_filter()
    temporal = rs.temporal_filter(0.4,20,7)
    hole_filling = rs.hole_filling_filter(2)
    #depth_to_disparity = rs.disparity_transform(True)
    #disparity_to_depth = rs.disparity_transform(False)
    colorizer = rs.colorizer()

    # Getting the depth sensor's depth scale (see rs-align example for explanation)
    depth_sensor = profile.get_device().first_depth_sensor()
    depth_scale = depth_sensor.get_depth_scale()
    print("Depth Scale is: " , depth_scale)

    # We will be removing the background of objects more than
    #  clipping_distance_in_meters meters away
    clipping_distance_in_meters = 1 #1 meter
    clipping_distance = clipping_distance_in_meters / depth_scale

    # Create an align object
    # rs.align allows us to perform alignment of depth frames to others frames
    # The "align_to" is the stream type to which we plan to align depth frames.
    align_to = rs.stream.color
    align = rs.align(align_to)

    # Streaming loop
    try:
        while True:
            # Get frameset of color and depth
            frames = pipe.wait_for_frames()
            # frames.get_depth_frame() is a 640x360 depth image

            # Align the depth frame to color frame
            aligned_frames = align.process(frames)

            # Get aligned frames
            aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
            color_frame = aligned_frames.get_color_frame()

            # Validate that both frames are valid
            if not aligned_depth_frame or not color_frame:
                continue

            depth_image = np.asanyarray(aligned_depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())

            # Remove background - Set pixels further than clipping_distance to grey
            grey_color = 153
            depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
            bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)

            # Render images:
            #   depth align to color on left
            #   depth on right
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
            images = np.hstack((bg_removed, depth_colormap))

            cv2.namedWindow('Align Example', cv2.WINDOW_NORMAL)
            cv2.imshow('Align Example', images)
            key = cv2.waitKey(1)
            # Press esc or 'q' to close the image window
            if key & 0xFF == ord('q') or key == 27:
                cv2.destroyAllWindows()
                break
    finally:
        pipe.stop()

d435realsense = Process(target=d435realsense, args=())
d435realsense.start()
```

MartyG-RealSense, 2021-06-19:
Hi @DieProgrammIDE   In a past Python case involving problems with align when using multiprocessing, it was advised to build librealsense with support for using multiple processor cores with the CMake build flag -**DBUILD_WITH_OPENMP=true**

https://github.com/IntelRealSense/librealsense/issues/4151

DieProgrammIDE, 2021-06-19:
"if you are able to run the script without exiting if you remove the line import face_recognition then it would suggest that the conflict is with the face_recognition module." 
That is exactly the case. I have compiled with OpenMP/CUDA I think, but Performance is not a problem when not importing "face_recognition".

I think it is some kind of conflict although I don't get any Errors which makes it hard for me to debug. Also I don't need to use face_recognition for this to happen, importing is enough. I also tried only having the Libraries Imported at the passages where they are used but it wasn't of any success either.

MartyG-RealSense, 2021-06-19:
There is a Python example of a multiprocessing script in the link below that you could try to see whether it works for you, and compare to your own script if it does work.

https://github.com/IntelRealSense/librealsense/issues/7365#issuecomment-694091530

A RealSense support team member has also suggested the pyrealsense2 multiprocessing example below:

https://gist.github.com/mengyui/8d00dbbefff65e70db9691965b99efb8

DieProgrammIDE, 2021-06-19:
> #7365 (comment)

Sorry, but to me it is very clear that the Problem isn‘t Multiprocessing. Anything Realsense is done in one process, so it shouldn‘t be a problem.

Everything works except aligning only when both libraries are imported

MartyG-RealSense, 2021-06-19:
So basically you are setting up support in your Python script for multiprocessing with **from multiprocessing import Process** but you are not using multiprocessing on the RealSense elements of your application?

I would be inclined to put everything under multiprocessing, or instead forget about using multiprocessing.  As you are using Jetson, you can get a significant performance gain when performing depth-color alignment due to librealsense's CUDA support of the Jetson's Nvidia graphics GPU.

closed 2021-07-03 by MartyG-RealSense