realsenseai/librealsense #9249: Module 'pyrealsense2' has no attribute 'pipeline'

Try-Hello, 2021-06-19:
I use pyrealsense2-aarch64 in jetson nano.     

import pyrealsense2 as rs
#And then define 'pipeline':
pipeline = rs.pipeline()

error：Module 'pyrealsense2' has no attribute 'pipeline'

MartyG-RealSense, 2021-06-19:
Hi @Try-Hello  There was a Jetson case involving **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** where the solution in that particular case was found to be to copy .so files to the same directory that the user's project script was in.

https://github.com/IntelRealSense/librealsense/issues/7078#issuecomment-672949007

Try-Hello, 2021-06-19:
Sorry, my knowledge in this area is very weak.
I don't know what the .so files are, and I don't know where the .so files are.

I installed pyrealsense2-aarch64 in this way：pip3 install  pyrealsense2-aarch64
where can i find the .so files？please。

Try-Hello, 2021-06-19:
> Hi @Try-Hello There was a Jetson case involving **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** where the solution in that particular case was found to be to copy .so files to the same directory that the user's project script was in.
> 
> [#7078 (comment)](https://github.com/IntelRealSense/librealsense/issues/7078#issuecomment-672949007)

Sorry, my knowledge in this area is very weak.
I don't know what the .so files are, and I don't know where the .so files are.

I installed pyrealsense2-aarch64 in this way：pip3 install pyrealsense2-aarch64
where can i find the .so files？please。

MartyG-RealSense, 2021-06-19:
I believe that the .so files are part of the source-code installation method and not involved in the pip packages.

I researched the pip3 install command that you used.  The **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** error has occurred in a past case when installing with the command **pip3 install pyrealsense2-aarch64**.  A RealSense team member advised uninstalling it and then installing with **pip 3 install pyrealsense2** instead.

https://github.com/IntelRealSense/librealsense/issues/4483

Try-Hello, 2021-06-23:
> I believe that the .so files are part of the source-code installation method and not involved in the pip packages.
> 
> I researched the pip3 install command that you used. The **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** error has occurred in a past case (see the link below) when installing with the command **pip3 install pyrealsense2-aarch64**. A RealSense team member advised uninstalling it and then installing with **pip3 install pyrealsense2** instead.
> 
> #4483

this problem has been solved. thank you very much.

MartyG-RealSense, 2021-06-23:
Great to hear, @Try-Hello - thanks very much for the update!

545464598, 2021-06-24:
> this problem has been solved. thank you very much.

hello,how did you solve this problem?

Try-Hello, 2021-06-24:
> hello,how did you solve this problem?

You can refer to this：The same as the official website introduction.
https://blog.csdn.net/weixin_39298885/article/details/113149462

closed 2021-06-30 by MartyG-RealSense