realsenseai/librealsense #9250: T261 lost tracking in certain direction when connected to a usb hub.

Dream-build, 2021-06-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T261 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  2.47.0 | 
| Language                            | unity| 
| Segment			|  AR/others | 

### Issue Description
<Describe your issue / question / feature request / etc..>
The module lost tracking and went straight down in the SDK viewer when plugin in a usb hub or my integrator board with Motion Leap hand tracking controller. I'm currently working on the project north-star mr headset with T261 and motion leap controller mounted on a usb hub or the integration board, it should be working perfectly as other North Star developers did but currently I'm not one of them. A usb hub and the integrator board were used and both resulted in tracking failure to certain direction when T261 and motion leap controller activated at the same time. However T261 is working perfectly when motion leap controller is not active but been plugged in.
Here's some of the recording and screen-shots.

[image]
[image]
https://drive.google.com/file/d/13_5_IpxWlGMCBuSmDjzla9LAqnM0kipf/view?usp=sharing

Dream-build, 2021-06-19:
More supporting material:
[image]
https://drive.google.com/file/d/13_5_IpxWlGMCBuSmDjzla9LAqnM0kipf/view?usp=sharing

MartyG-RealSense, 2026-10-25:
This open issue relating to the RealSense T261 / T265 camera is being closed because the T265 model is End of Life (EOL) and no longer supported. We welcome the creation of new issues relating to supported cameras.

closed 2026-10-25 by MartyG-RealSense