realsenseai/librealsense #9251: High-speed mode save image mode

Ligcox, 2021-06-19:
[errorexample.txt](https://github.com/IntelRealSense/librealsense/files/6680965/errorexample.txt)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |05.12.13.50| 
| Operating System & Version |   Win (8.1/10)   | 
| Kernel Version (Linux Only)    | -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.25.0                          | 
| Language                            |  python                           | 
| Segment			|  others                          | 

### Issue Description
I tried to use the D435i high-speed mode(<https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435>) to record the image data and write them to the .npz file. The program ran into problems when I tried to add data to a variable in the for loop. I tried the following

1. Comment an invalid operation statement, such as using OpenCV to display colormap
2. Load data with an empty list and add data to the list using methods append or insert, for example,  res=[]; res.append(depth_image)
3. Read the image data through two separate threads and add it to the final result
4. Create a FIFO queue using queue.Queue() and return the res
5. Change the different wait times in pipeline.wait_for_frames()

Run into the same problem using these methods, they're similar

Traceback (most recent call last):
File ".\src\realsense\save_data.py", line 17, in <module>
  frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

I made sure that my RealSense was connected to the computer correctly, and I used the original data cable. RealSense work fine in the Intel RealSense Viewer and are displayed over the USB3.2 cable connection.

When I comment out the code that saves the image, the error goes away (for example, comment line 28 of the instance file). This error occurs during the 15th for loop, and you can see through the debuger that the program works normally during the previous loops.

The error demonstration code is as follows

        import pyrealsense2 as rs
        import cv2
        import numpy as np
        import time
        import tqdm
        
        pipeline = rs.pipeline()
        config = rs.config()
        config.enable_stream(rs.stream.depth, 0, 848, 100, rs.format.z16, 300)
        pipeline.start(config)
        
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        num = 1000
        res = [None for i in range(num)]
        
        for i in tqdm.tqdm(range(100)):
            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            if not depth_frame:
                continue
        
            depth_image = np.asanyarray(depth_frame.get_data())
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(
                depth_image, alpha=0.085), cv2.COLORMAP_JET)
            cv2.imshow('RealSense', depth_colormap)
            cv2.waitKey(1)
        
            res[i] = depth_image
        
        cv2.destroyAllWindows()
        np.savez_compressed("realsense", *res)

MartyG-RealSense, 2021-06-19:
Hi @Ligcox  When an **append** operation in a Python script in librealsense fails after the 15th frame, it is typically due to a limitation on saving frames into a collection.  

https://github.com/IntelRealSense/librealsense/issues/946#issuecomment-353878904

A solution that has worked for RealSense users who experience this issue is to store the frames in memory using the SDK's **Keep()** function.  The links below have examples of Python code for Keep():

https://github.com/IntelRealSense/librealsense/issues/3164#issuecomment-458959674

https://github.com/IntelRealSense/librealsense/issues/3121#issuecomment-466371013

MartyG-RealSense, 2021-06-28:
Hi @Ligcox  Do you require futher assistance with this case, please?  Thanks!

closed 2021-07-04 by MartyG-RealSense