realsenseai/librealsense #9259: RealSense runs for a long time with no input stream

Rachel1111, 2021-06-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430                                                       | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18                                            | 
| Kernel Version (Linux Only)    | 4.4.194                                                   | 
| Platform                                 | Ubuntu                                                  |
| SDK Version                            |  2.0                                                        | 
| Language                            |  c++                                                         | 
| Segment			|  others                                                           | 

### Issue Description
hello, I found an issue. RealSense cannot get  new input streams if it runs for a long time , just stops at 'rs2::frameset frame_data = pipe_.wait_for_frames()'. So, how can I solve it.

MartyG-RealSense, 2021-06-21:
Hi @Rachel1111   An error that has been known to occur with long-run sessions of a RealSense camera is **RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x21ad1b0): Frame didn't arrive within 15000**.  Do you receive an error message like this one when the problem occurs, please?

Rachel1111, 2021-06-21:
No, I don't receive this error. It stops before frame_data = pipe_.wait_for_frames().
[image]

MartyG-RealSense, 2021-06-21:
Thanks very much.  Is **RealSenseDriver.cpp** the name of your project's script that you have created?

Can you also confirm the RealSense camera model that you have please.  The information at the top of the case lists the **D430** model, whilst the above image says **SR300**.

Rachel1111, 2021-06-21:
Thanks a lot. The realsense I use is D430, SR300 is the realsense I used before and I forgot to correct.

Rachel1111, 2021-06-21:
Yeah,  RealSenseDriver.cpp is the name of my project's script .

MartyG-RealSense, 2021-06-21:
Is '90' a value that you have deliberately set the **Laser Power** option to, or is it the current value that you are reading from the camera at the time that the problem occurs?

Rachel1111, 2021-06-21:
'90' is a value I set the Laser Power option, and 7500 is also the value I set Exposure option.

MartyG-RealSense, 2021-06-21:
Thanks for the confirmation.  Could you also check the camera ASIC Temperature and the Projector Temperature under the **Controls** section of the Stereo Module options and see what the temperature values are after a few minutes of running the camera.

Rachel1111, 2021-06-21:
Thanks a lot. yes, I have a try to check the camera ASIC Temperature and  Projector Temperature later on, and see these values.

MartyG-RealSense, 2021-06-27:
Hi @Rachel1111  Do you have an update about this case that you can provide, please?  Thanks!

Rachel1111, 2021-06-28:
Sorry to reply you so late. I have updated the Firmware to Signed_Image_UVC_5_12_14_50.bin, but the D430 also had the question that it cannot get the stream and wait at pipe_.wait_for_frames()
[image]

Rachel1111, 2021-06-28:
And I checked the camera ASIC Temperature and Projector Temperature, them seemed not changed a lot.

MartyG-RealSense, 2021-06-28:
The sense that I get from the above comments is that the program is freezing, requiring you to cancel out of it with Ctrl-C.  Is that correct, please?

If the program is frozen then this could account for why you do not receive 'time-out' errors such as **Frames not received within 15000** ... because the program is unable to update and detect a time-out due to its frozen state.

Could you check whether a 'memory leak' might be occurring in your program, where the available memory capacity reduces over time until there is no longer enough memory for the program to continue running.  In Ubuntu you can monitor memory consumption using a tool such as **htop**.

https://vitux.com/how-to-use-htop-to-monitor-system-processes-in-ubuntu/

Rachel1111, 2021-06-29:
Hi, I just monitored memory consumption via htop, and the memory the program used seemed not to account for a half.

MartyG-RealSense, 2021-06-29:
Your situation reminds me of the recent case https://github.com/IntelRealSense/librealsense/issues/9184 in which the pipeline was freezing without any error message after using **sensor.Close()**.  Does your own application make use of this instruction to stop the sensor, please?

Rachel1111, 2021-06-30:
yeah, I used pipeline.stop(),and then I restart it in my code.
[image]

MartyG-RealSense, 2021-06-30:
Some RealSense users have dealt with problems involving pipeline.Stop() by using pipeline.Start() and **sensor.**Stop()

https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

In other words.  pipeline start, sensor stop, pipeline start, etc

SDK **2.48.0** is also released now, so it may be worth testing your program with that new SDK version if your project is able to change versions.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

Rachel1111, 2021-07-02:
Thanks for your suggestion, and I would try it later.

Rachel1111, 2021-07-08:
I'm so sorry to tell you that the problem still exist after I updated the SDK to v2.48.0

MartyG-RealSense, 2021-07-08:
There is no need to apologize, it's no trouble at all.  :)

Were you able to try the earlier suggestion in https://github.com/IntelRealSense/librealsense/issues/9259#issuecomment-871158337 to use pipeline start and **sensor** stop?

MartyG-RealSense, 2021-07-14:
Hi @Rachel1111  Do you require further assistance with this case, please?  Thanks!

Rachel1111, 2021-07-18:
Hi, @MartyG-RealSense ,I'm sorry to reply so late. Actually, I have found another problem when I updated my SDK to 2.48. The Depth data doesn't change sometimes, they just maintain in the first stream when the camera started. I mean that these depth images I saved are the same.

MartyG-RealSense, 2021-07-18:
Do you experience the same behavior in the RealSense Viewer or only with your project script, please?

Rachel1111, 2021-07-19:
Hi, @MartyG-RealSense ,I seemed to solve the problem by start only depth stream. 
[image]

MartyG-RealSense, 2021-07-19:
Thanks very much @Rachel1111 for the update.   Your application still needs an infrared stream to be enabled for it to work as intended though, right?

Rachel1111, 2021-07-28:
Thanks, @MartyG-RealSense I don't have any questions now.

MartyG-RealSense, 2021-07-28:
Thanks very much @Rachel1111 for the update!  I will close this case now.

closed 2021-07-28 by MartyG-RealSense