realsenseai/librealsense #9263: [Question] Extrinsic parameter of D415 color frame

wodndb, 2021-06-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |  Win10 21H1  | 
| Platform                                 | PC/Unity(2020.3.9f LTS) |
| SDK Version                            | 2.41.0 | 
| Language                            | C#/Unity | 
| Segment			| others | 

### Issue Description
I try to texturing 3D mesh which is reconstructed using D415 cameras.

For texturing, I need extrinsic parameter of D415 color frame.

In datasheet, distance of right IR image sensor and color image sensor is 15 mm.

So I set extrinsic parameter of color frame like this.

[image]

However, color image is not aligned to mesh.

Can I get extrinsic parameter of color frame?

MartyG-RealSense, 2021-06-22:
Hi @wodndb  You can obtain the extrinsic and intrinsic parameters of your particular camera by running the **rs-enumerate-devices** tool in the Windows Command Prompt interface in its calibration information mode by following the steps below:

**STEP 1**
Open the Command Prompt interface.  you can find it by typing **command** into the text box at the bottom of your Windows screen, next to the corner button where the PC is shut down.

**STEP 2** 
Copy and paste the command below into the Command Prompt interface to navigate to the Tools directory where the rs-enumerate-devices tool should be located.

**cd C:\Program Files (x86)\Intel RealSense SDK 2.0\tools**

**STEP 3**
Input the command below to launch rs-enumerate-devices in calibrtion information mode.

**rs-enumerate-devices -c**

This command will generate a long list of extrinsic and intrinsic data.  You can then scroll up through the listing in the terminal until you reach the category of information that you require.

If you need details of the extrinsics between the right IR sensor (Infrared 2) and the color sensor then the category **Extrinsics from "Infrared 2" to "Color"** may be what you require.

Here are the results from this category from my own D415 camera.

[image]

However, since the origin of the coordinate system of 400 Series cameras is the **left** infrared sensor, the category **Extrinsics from "Infrared 1" to "Color"** may be more suitable.

[image]

wodndb, 2021-06-22:
@MartyG-RealSense Thanks a lot, I try to use rs-enumerate-devices tool.

wodndb, 2021-06-23:
First, I get depth aligned color frame which is generated from post processing pipe and texturing. It works well.
[image]

And I get raw color frame which is streamed from rsDevice and texturing with color to depth extrinsics. However this extrinsic is not good. texture is not aligned.
[image]

In my opinion, there are some other extrinsic for color frame to depth aligned color frame.

MartyG-RealSense, 2021-06-23:
There is an example scene in the RealSense Unity wrapper called **AlignmentSample** that demonstrates usage of the alignment processing block by using the wrapper's **RsAlign** script.  Have you tried this example scene please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsAlign.cs

[image]

wodndb, 2021-06-25:
@MartyG-RealSense

I use three D415 cameras and I want to use 2K color image to texturing. If I use rs align for depth to color alignment, post processing block overhead will be increased.

So I can't use rsalign.

wodndb, 2021-06-25:
I think that depth to color camera extrinsic parameter and color to depth camera extrinsic parameter are not working.

Also extrinsic from rs-enumerate-devices tool and extrinsic from RealSense SDK method in Unity is not same.

I don't know why.

I get camera extrinsic and intrinsic parameter like this in RealSense SDK for Unity.

### RsPointCloudRenderer.OnStartStreaming
```
(...)
            using (var depth = obj.Streams.FirstOrDefault(s => s.Stream == Stream.Depth && s.Format == Format.Z16)
                .As<VideoStreamProfile>())
            {
                ResetMesh(depth.Width, depth.Height);
                var intrinsics = depth.GetIntrinsics();
                var extrinsics = depth.GetExtrinsicsTo(obj.Streams.FirstOrDefault(s => s.Stream == Stream.Color));
            }
(...)
```

And I find interesting problem. After post processing pipe with RsAlign option, intrinsic parameter of color frame is not changed although color frame is aligned to depth frame.

MartyG-RealSense, 2021-06-25:
I studied your problems carefully.  Using **rs-align** worked for you, but the slowdown from performing alignment with three cameras was unsatisfactory.  I wonder whether building librealsense from source code with the **BUILD_WITH_OPENMP** build flag set to true to add support for using multiple processor cores might speed up alignment and color conversion operations.  The trade-off with using that method may be a higher percentage usage of the CPU.

[image]

In regard to your concerns about the intrinsic parameter of the color frame apparently not changing: I am not certain about this question, though I bear in mind that the rendering of camera data in Unity is not quite the same as when using other wrappers such as C++, Python and C#.  In the Unity wrapper, camera data is rendered visibly as a **material** rather than direct digital values.

wodndb, 2021-06-28:
@MartyG-RealSense Thanks to reply.

I don't know how to build librealsense to Unity Wrapper. So it's hard to use openmp.

I just try to projection 2K image to 3D model which is reconstructed from captured point cloud.

But I failed to texture alignment.

So I think that rs align algorithm is not just projection some frame to other frame.

I need to study rs-alignment and develop rs align for just one frame.

closed 2021-06-28 by wodndb

wodndb, 2021-06-28:
@MartyG-RealSense 

I test ICP registration between original point cloud and color frame aligned point cloud using [Open3D](http://www.open3d.org/docs/release/tutorial/pipelines/colored_pointcloud_registration.html).
And transform matrix is differ to extrinsic of color frame.

### Before registration
[image]

### After registration
[image]

### Transform Matrix
[image]

I think that extrinsic parameter and some other transform matrix which I don't know are used for rs align.

MartyG-RealSense, 2021-06-28:
When aligning depth to color in the RealSense SDK, the **align processing block** is used to resize the field of view size automatically.  When performing depth to color alignment, the depth FOV size is resized to the color FOV size.  If color is being aligned to depth then the color FOV size is resized to the depth FOV size.

wodndb, 2021-07-01:
@MartyG-RealSense Thanks to reply.

I try to write some codes for custom color-depth alignment, but it's hard to me.

So I change some option of rs devices to process rs align for high resolution streaming images from multi D415 camera.

1. I changed process mode of rsDevice to `multithread`.
2. I changed frame rate of color and depth from to `15`.

#### Unity Profiler (Unity Thread)
[image]

#### Unity Profiler (Multithread)
[image]

Multithread mode is good for processing multi camera streaming in Unity.

Although rs align is enabled for three depth and color image, frame rate maintain 60 in multithread mode lol.

But I can't use my computer vision algorithm to streaming image in real time because resolution of image is too large.

Anyway, I decide to use rs align using multi thread mode of rsDevice.