realsenseai/librealsense #9265: crashing as soon as toggling the streaming button

gilles-syrto, 2021-06-22:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2021-06-22:
Hi @gilles-syrto  Could you try downloading the standalone version of the Viewer called **Intel.RealSense.Viewer.exe** from the 'Assets' file list on the SDK Releases page and see whether the error still occurs on that Viewer version, please? 

This Viewer should be launched with a double-leftclick from the folder that it is downloaded to, and not with the Viewer launch icon on the Windows desktop.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

[image]

If using the standalone Viewer does not solve the problem then please next try disabling the two **GLSL** options in the Viewer's **Settings** interface using the instructions in the link below.

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

MartyG-RealSense, 2021-06-29:
Hi @gilles-syrto  Do you require further assistance with this case, please?  Thanks!

gilles-syrto, 2021-06-30:
Hello, 

 �

Sorry I’ve been busy with some other stuff. I updated the FW of the lidar and got the SDK, and it is now fine… sorry for the late answer and thanks for your help.

 �

Best regards,

Gilles

 �

 �

MartyG-RealSense, 2021-06-30:
No problem at all @gilles-syrto - that's great to hear that your problem is resolved.  I will close the case as you have achieved a solution.  Thanks for the update!

closed 2021-06-30 by MartyG-RealSense