realsenseai/librealsense #9272: Questions about the use of D435

ju-mingyue, 2021-06-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435  }                                                  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu18.04.5)                            | 
| Kernel Version (Linux Only)    |  (4.9.201)                                                | 
| Platform                                 | NVIDIA JetsonNano B01                        |
| SDK Version                            |  2.41                                                      | 
| Language                            |  ROS packages                                          | 
| Segment			|                  Robot                                            | 

### Issue Description
Hello, I encountered the following problems when using the D435 camera:
1. When I run the rs_rgb.launch file in the ros package, the system prompts the following error:
    catkin_ws$ roslaunch realsense_ros_camera rs_rgbd.launch 
... logging to /home/hanning/.ros/log/7461e208-d3c9-11eb-b336-00044be5a932/roslaunch-hanning-desktop-11286.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
ResourceNotFound: rgbd_launch
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/hanning/robot_ws/src
ROS path [2]=/opt/ros/melodic/share
2. When I run D435 related program, I was able to view / camera / depth / camera_info topic name, but when I use rostopic echo / camera / depth / camera_info to view specific content publishing when, but no message content. But rostopic echo /camera/color/camera_info can see the message content.
3. I am not sure how to use the depthimage_to_laserscan package to do visual obstacle avoidance. Hope to provide some suggestions or references. Thank you very much.

MartyG-RealSense, 2021-06-23:
Hi @ju-mingyue 

1.  Support for the **rs_rgbd** launch method on ROS1 Melodic should be installed using the instruction below.

**sudo apt install ros-melodic-rgbd-launch**

2.  After installing support for the rs_rgbd relaunch instruction, could you repeat the launch please.  If launch completes successfully, use the command **rostopic list** to confirm whether the topic **camera/depth/camera_info** is available.  If it is available, perform the rostopic echo of this topic to see whether it now provides information.  

If it still does not provide information, please copy and paste the text of your roslaunch log into this discussion.

3.  If you have not been able to launch **depthimage_to_laserscan**, a guide in the link below for doing so with D435 and the RealSense ROS wrapper may be helpful.

https://programmersought.com/article/62234275154/

If you are able to launch it successfully, the RealSense ROS forum link below discusses use of depthimage_to_laserscan for mapping with D435.

https://github.com/IntelRealSense/realsense-ros/issues/510

My research indicates that this package tends to be used with a D435 and a T265 Tracking Camera in combination though.

https://github.com/IntelRealSense/realsense-ros/issues/933
https://intel.github.io/robot_devkit_doc/pages/rs_slam.html

A useful reference for achieving obstacle avoidance with a D435 may be the Intel-provided YouTube tutorial in the link below about using a D435 and T265 for navigation and avoidance.  Even if you do not have a T265, the D435 aspect of the project that is demonstrated may be helpful.  The video talks about the D435 at the 27 minutes 25 seconds point.

https://m.youtube.com/watch?v=62vm0_RZ1nU

ju-mingyue, 2021-06-24:
> 嗨@ju-mingyue
> 
> 1. 应使用以下说明安装对 ROS1 Melodic 上的**rs_rgbd**启动方法的支持。
> 
> **须藤 apt 安装 ros-melodic-rgbd-launch**
> 
> 1. 安装对 rs_rgbd 重新启动指令的支持后，请您重复启动。如果启动成功，使用命令**rostopic list**确认主题**camera/depth/camera_info**是否可用。如果可用，则执行该主题的 rostopic echo 以查看它现在是否提供信息。
> 
> 如果它仍然没有提供信息，请将您的 roslaunch 日志文本复制并粘贴到此讨论中。
> 
> 1. 如果您无法启动**depthimage_to_laserscan**，以下链接中有关使用 D435 和实感 ROS 包装器执行此操作的指南可能会有所帮助。
> 
> https://programmersought.com/article/62234275154/
> 
> 如果您能够成功启动它，下面的 RealSense ROS 论坛链接讨论了使用 depthimage_to_laserscan 与 D435 进行映射。
> 
> [IntelRealSense/realsense-ros#510](https://github.com/IntelRealSense/realsense-ros/issues/510)
> 
> 我的研究表明，该套件倾向于与 D435 和 T265 跟踪相机结合使用。
> 
> [IntelRealSense/realsense-ros#933 ](https://github.com/IntelRealSense/realsense-ros/issues/933)
> https://intel.github.io/robot_devkit_doc/pages/rs_slam.html
> 
> 英特尔过去曾为 ROS发布了一个**ros_object_analytics**包，其中包括智能避障，尽管它现在已经很老了，并且是为现已过时的**ROS Kinetic 设计的**。当与实感 400 系列相机一起使用时，它依赖于**rs_rgbd.launch**
> 
> https://github.com/intel/ros_object_analytics

 Thank you very mutch! 
Now I need to use the obstacle avoidance function in the process of using D435. At present, there are two ways I have inquired:

Use depthimage_to_laserscan to convert the depth information into a lidar signal, but the current problem is that there is also a lidar on my robot. Now these two topics are scan, so there is a conflict. I don't know how to solve it.

I want to know whether the two lidar signals can be fused, and what configuration is needed to be fused. Is there any relevant information or code?

Using plotcloud2 point cloud information, I don't understand how to do this at present. Although the point cloud image can be seen on the map now, it does not have the effect of avoiding obstacles. And does this point cloud information need to be passed to AMCL? If so, how does it need to be delivered? So I hope someone can help me.

MartyG-RealSense, 2021-06-24:
@ju-mingyue  Thanks very much.  I will reply to the above question at the new issue https://github.com/IntelRealSense/librealsense/issues/9282 where you have asked it.

ju-mingyue, 2021-06-29:
@MartyG-RealSense
 Hello, regarding the use of the rgbd.launch file, the following problem occurred after running, but the in-depth topic can be published normally.
[image]

MartyG-RealSense, 2021-06-29:
Thanks very much @ju-mingyue for the information about the power_line warning.  I do not think that it is significant, and you have found yourself that the topics are publishing correctly.

closed 2021-07-02 by ju-mingyue