realsenseai/librealsense #9277: Failed to set power state

UserFLO, 2021-06-23:
Hi ! I use Camera D455 and i have this issue when i run my program. I'm on Raspberry pi. Do you have a solution to fix it ? Thanks ! 
[image]

MartyG-RealSense, 2021-06-23:
Hi @UserFLO  Could you provide information please about the questions below

-  Are you using Raspbian OS or Ubuntu
-  Which Raspberry Pi model are you using (3B, 4, etc)
-  Are you using a single D455 camera or more than one camera

UserFLO, 2021-06-23:
I am using Raspbian OS with a 3B RPI . And i have just one D455 Camera

MartyG-RealSense, 2021-06-23:
Thanks very much.  So the camera would be on a USB 2.0 connection, as Pi models before the Pi 4 did not have USB 3.  What stream resolution and FPS are you using in your program, please?

Also, what stream types are you using (depth, color, infrared, etc).

UserFLO, 2021-06-23:
`using namespace std; 

typedef pcl::PointXYZRGB RGB_Cloud;
typedef pcl::PointCloud <RGB_Cloud> point_cloud; 
typedef point_cloud:: Ptr cloud_pointer;
typedef point_cloud:: Ptr prevCloud; 

void Load_PCDFile(void);
bool userInput(void); 
void cloudViewer(void);

string cloudfile;
string prevCloudFile;
//int i =1; 

std::tuple<int,int,int> RGB_Texture(rs2::video_frame texture, rs2::texture_coordinate Texture_XY) 
{
	int width = texture.get_width(); 
	int height = texture.get_height();
	
	int x_value = min(max(int(Texture_XY.u * width + .5f),0), width-1); 
	int y_value = min(max(int(Texture_XY.v * height + .5f),0), height-1); 

	int bytes = x_value * texture.get_bytes_per_pixel(); 
	int strides = y_value * texture.get_stride_in_bytes(); 
	int Text_Index = (bytes + strides); 
	
	const auto New_Texture = reinterpret_cast<const uint8_t*>(texture.get_data()); 
	
	int NT1 = New_Texture[Text_Index]; 
	int NT2 = New_Texture[Text_Index +1]; 
	int NT3 = New_Texture[Text_Index +2]; 
	
	return std::tuple<int,int,int>(NT1,NT2,NT3); 
} 

cloud_pointer PCL_Conversion(const rs2::points&points, const rs2::video_frame &color){

	cloud_pointer cloud(new point_cloud); 
	std::tuple<uint8_t, uint8_t,uint8_t> RGB_Color; 
	
	auto sp = points.get_profile().as<rs2::video_stream_profile>();
	
	cloud->width = static_cast<uint32_t>( sp.width() );
	cloud->height = static_cast<uint32_t>( sp.height() );
	cloud->is_dense = false; 
	cloud->points.resize( points.size() ); 
	
	auto Texture_Coord = points.get_texture_coordinates(); 
	auto Vertex = points.get_vertices(); 
	
	for (int i =0; i<points.size(); i++)
	{
		cloud->points[i].x = Vertex[i].x; 
		cloud->points[i].y = Vertex[i].y; 
		cloud->points[i].z = Vertex[i].z; 
		
		RGB_Color = RGB_Texture(color, Texture_Coord[i]); 
		
		cloud->points[i].r = get<2>(RGB_Color); 
		cloud->points[i].g = get<1>(RGB_Color);
		cloud->points[i].b = get<0>(RGB_Color);
		
	} 
	return cloud;
}

void acquisition (pcl::PointCloud<pcl::PointXYZ>::Ptr& cloud_out, int cptcapt)
{
	bool captureLoop = true; 
	
	pcl::PointCloud<pcl::PointXYZRGB>::Ptr newCloud (new pcl::PointCloud<pcl::PointXYZRGB>); 

	boost::shared_ptr<pcl::visualization::PCLVisualizer> openCloud; 
	
	rs2::pointcloud pc; 
	rs2::pipeline pipe; 
	rs2::config cfg; 
	
	cfg.enable_stream(RS2_STREAM_COLOR); 
	cfg.enable_stream(RS2_STREAM_INFRARED);
	cfg.enable_stream(RS2_STREAM_DEPTH);
	
	
	rs2::decimation_filter dec_filter;
    
    dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 4);
    
	
	rs2::pipeline_profile selection = pipe.start(cfg); 
	
	
	
	auto frames = pipe.wait_for_frames(); 
	auto depth = frames.get_depth_frame();
	auto color = frames.get_color_frame(); 
	
	pc.map_to(color); 
	 
	rs2::frame filtered = depth;
    filtered = dec_filter.process(filtered);
	auto points = pc.calculate(depth); 
	
	cloud_pointer cloud = PCL_Conversion(points,color);
	pcl::copyPointCloud(*cloud,*cloud_out);
	pcl::io::savePCDFile ("cloudXYZ.pcd", *cloud_out);
	
	
	//pcl::copyPointCloud(*cloud, *cloud_out);
	/*pcl::PassThrough<pcl::PointXYZRGB> Cloud_Filter; 
	Cloud_Filter.setInputCloud(cloud); 
	Cloud_Filter.setFilterFieldName("z");
	Cloud_Filter.setFilterLimits(0.0,1.0); 
	Cloud_Filter.filter(*newCloud); */ 
	
	//cloudfile = "Captured_Frame" + to_string(i) + ".pcd"; 
	cloudfile = "Captured_Frame_carton.pcd"; 
	cout << "Generating PCD Point Cloud File..." <<endl; 
	pcl::io::savePCDFileASCII(cloudfile, *cloud_out); 
	cout<< cloudfile << "successfully generated."<< endl; 
	
	
	
	cout <<"Exiting Program..." <<endl; 
	//return EXIT_SUCCESS;
} `

I attach you a part of my program.

MartyG-RealSense, 2021-06-23:
As the Pi 3B is a low-specification board (1 GB memory and USB 2), applying decimation post-processing may be over-burdening the board as post-processing is calculated in the CPU of the computer / computing device and not on the camera's internal hardware.  

Also, are you setting up the configuration of the camera and its start in a loop?  If so, these instructions should not be looped.  They should only be run once when the program is launched.

[image]

Once the pipeline has been started then it cannot be started again until it is first stopped, otherwise a **Failed to set power state** could result because the pipeline is in a 'busy' state.

UserFLO, 2021-06-25:
Hi @MartyG-RealSense ! I followed your instructions. I run just one time the configuration. But now, i obtain this error : 
[image]
Thanks !

MartyG-RealSense, 2021-06-25:
**wait_for_frames()** should be placed just after the pipeline start and not before it, like in your earlier script.

[image]

UserFLO, 2021-06-25:
Ok i understand ! And where **dec_filter.set_option(RS2_OPTION_FILTER_MAGNITUDE,, 4)** should be placed ?

MartyG-RealSense, 2021-06-25:
In the majority of scripts that I researched where **RS2_OPTION_FILTER_MAGNITUDE** was defined, it was placed after the pipeline start.

closed 2021-07-07 by MartyG-RealSense