realsenseai/librealsense #9278: IMU hangs on close (Unity3D)

Vangos, 2021-06-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i & D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |  -                                    | 
| Platform                                 | PC |
| SDK Version                            |  2.45.0 }                          | 
| Language                            |  C#/Unity3D                          | 
| Segment			|  Computer application                         | 

## Issue Description

If the IMU (accelerometer and/or gyroscope) is enabled in Unity3D, the application hangs for about 20 seconds when closing.

### Unity Editor
The camera starts properly and works fine. I can access the color, depth, and IMU data. However, when I click the Stop button in Unity3D, the Editor hangs for about 20 seconds.

### Unity Build
If I export the project as a Windows standalone executable, the executable also works fine. However, when I click the Close X button, the window hangs for about 20 seconds.

## Steps to reproduce

- Create an empty Unity project and import the official RealSense Unity package.
- Open the `TexturesDepthAndColor` sample.
- In the `RSDevice` Editor field, add the `Accel` and `Gyro` streams.
- Click Play.
- Notice that the demo works fine.
- Click Stop.
- Hangs!

Here's the Editor configuration:

[image]

Vangos, 2021-06-23:
### Execution time

I've measured the execution time when disposing of the pipeline in the `RsDevice.cs` file:

```C#
var sw = new System.Diagnostics.Stopwatch();
sw.Start();

m_pipeline.Dispose();
m_pipeline = null;

sw.Stop();

Debug.Log($"Elapsed milliseconds: {sw.ElapsedMilliseconds}");
```

If the IMU is enabled, the code above prints **21057 milliseconds**. If the IMU is not enabled, the code prints **1072 milliseconds**.

MartyG-RealSense, 2021-06-23:
Hi @Vangos  This issue with a 20 second delay when closing a Unity session that had been using the IMU has been reported before in 2020.  

https://github.com/IntelRealSense/librealsense/issues/5993

At the time, the possibility that it was a bug was considered but as there were no further reports of the issue from other RealSense users, no further action was taken.

The RealSense user who created that case provided a method for recreating the issue.

https://github.com/IntelRealSense/librealsense/issues/5993#issuecomment-598290523

As more than one person has now reported this problem, I will seek advice from Intel about it.  Thanks for the report!

Vangos, 2021-06-23:
Thanks for your response. Just to address potential questions:
- Using the latest SDK (2.45.0).
- Using the latest firmware.
- No custom calibration.

MartyG-RealSense, 2021-06-23:
Thanks again @Vangos - I will update here when I have received a response from Intel.

MartyG-RealSense, 2021-06-24:
Hi @Vangos  After discussion with Intel, I have filed an official feature request with Intel to investigate this IMU time-delay issue with the Unity wrapper.  This case should be kept open whilst the feature request is active.  You do not need to take further action though.  Thanks for the report!

Vangos, 2021-06-24:
Thank you very much :-)

customlogic, 2021-07-13:
I have also experienced this problem as described, using a D455. 
An even easier way to reproduce it is to remove all streams from the RsDevice "Profiles" list. This causes the device to default to sending all 4 streams (Color, depth, gyro, accel), and you'll see the 20 second hang when you stop the editor.

Vangos, 2021-11-17:
Version 2.49 has resolved this issue for me, both regarding the Unity Editor and reading a standalone build.

Tested on Windows, macOS, and Linux. Works fine!

**Anyone facing this issue should update to version 2.49.**

@MartyG-RealSense Thank you for your support!

MartyG-RealSense, 2021-11-17:
Hi @Vangos  It's excellent to hear that SDK 2.49.0 resolved your problem!

As there is an active internal Intel report involving IMU delay in Unity that is linked to this case, the case should be kept open until that report is concluded.  You do not need to take any further action though, and I will update Intel about your success.  Thanks again!

BlueprintBen, 2021-11-17:
I tried updating the Unity plugin to 2.49.0 but am still seeing the issue unfortunately, at least in the Editor:
[image]

MartyG-RealSense, 2021-11-18:
Hi @BlueprintBen   Can you confirm whether you updated the Unity files to 2.49.0 by downloading and running the **Intel.RealSense.unitypackage** file from the 'Assets' file list on the SDK 'Releases' page, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

[image]

If you did use this Unity package and the delay on exit remained, would it be possible to upgrade to the new SDK **2.50.0** and similarly use the Unity package from its Assets list to see whether it makes a positive difference?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

MartyG-RealSense, 2021-11-18:
Thanks very much @BlueprintBen for your report.  I have updated the internal Intel report about this issue with details of your experience of delay in 2.49.0 and 2.50.0.  There is no further advice that can be provided about the issue at this time.

MartyG-RealSense, 2026-10-25:
Case closed due to no further comments received and the Unity platform no longer supported by RealSense.

closed 2026-10-25 by MartyG-RealSense