realsenseai/librealsense #9280: NodeJs wrapper Save to Disk script time out waiting for frames

chandujpc123, 2021-06-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, I'm facing frames not received in 5000 ms error when I'm trying to run two scripts (one for capture) and one for save using the available examples. Please let me know how to resolve this error.
Steps to reproduce:
1. In one command prompt, I am running nodejs-capture.js. The Viewer opens fine and shows color and depth frames on the viewer.
2. While above is running, I want to run save. I am doing this to live feed the stream and also save in parallel of the visible frame. So in another terminal, I launch the nodejs-save-to-disk.js program. After few seconds, I get the below error.
   **UnrecoverableError: Unrecoverable! error native function rs2_pipeline_wait_for_frames: Frame didn't arrive within 5000**

I can run nodejs-capture.js and nodejs-save-to-disk.js one after the other but not in parallel in separate terminals. Please let me know how can i get these two programs running in parallel.

MartyG-RealSense, 2021-06-23:
Hi @chandujpc123  The two scripts are operating independently, so both are trying to access the RGB stream.  Under the rules of sensor access described in the **Multi-Streaming Model** documentation in the link below though, only one process can 'claim' a particular stream on a particular camera.  Another application trying to access that particular stream will be unable to do so until the first application releases its claim on the stream.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

If you are generating a live-stream first with **nodejs-capture.js** and then running **nodejs-save-to-disk.js** secondly to capture an RGB snapshot to file then nodejs-save-to-disk.js would be unable to initiate the capture because nodejs-capture.js has already claimed exclusive access to the RGB stream.

What may work better is to add a keypress listener routine to nodejs-capture.js that listens for the press of a certain key, such as 's' for save, and then activates an image file export if a press of that key is recognized.  By having the image save routine embedded within nodejs-capture.js then only one application will be claiming the RGB stream.

chandujpc123, 2021-06-24:
Hi @MartyG-RealSense ,
Thanks for your response. It resolved my problem up to some extent - able to save the images per your suggestion.
But the same does not work if I try to open Stream(frame) to create .bag file. Any clues on this please?
Also, can the Intel RealSense Viewer play back the .bag file? I don't see any option to do so on the viewer. Please advice on this as well as I am a learner in this field.

MartyG-RealSense, 2021-06-24:
I'm pleased to hear that you are able to save images now.

Are you using the Nodejs Viewer in **nodejs/examples/realsense_viewer/** please?

chandujpc123, 2021-06-24:
yes, I am able to launch the web viewer using the nodejs/examples/realsense_viewer example. At the same time, I am trying to enhance it to save in .bag for whatever is shown on the web

MartyG-RealSense, 2021-06-24:
I researched your question carefully.  The wrapper Nodejs wrapper in librealsense wraps C++, ad it is possible to record a bag in C++ using the **enable_record_to_file** instruction, like the C++ code snippet in the link below.

https://github.com/IntelRealSense/librealsense/issues/2749#issuecomment-441629786

My research could not find any references for accessing bag recording from the NodeJs wrapper though unfortunately.

chandujpc123, 2021-06-24:
Thanks @MartyG-RealSense I'll try to do more research using the references you've shared. I think we are good to close this ticket for now.

closed 2021-06-24 by chandujpc123