realsenseai/librealsense #9281: Build fails for linux after v2.44

bontsai, 2021-06-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
This is an issue with building the source code in linux.  Using Linux Ubuntu 20.04, the last known good build seems to be v2.43.  It seems this build error is in every version after v2.43 I tried to compile v2.44, v2.45, v2.47 and they all give this build error:

Using gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)

When I tried to build with these cmake settings:

cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DCMAKE_INSTALL_PREFIX=/opt/realsense

There is an issue with what seems to the depth quality.  It fails with this error:

The first error is:

<pre>n file included from <b>/home/richardt/projects/apira-avatar/librealsense/common/realsense-ui-advanced-mode.h:7</b>,
                 from <b>/home/richardt/projects/apira-avatar/librealsense/common/model-views.h:26</b>,
                 from <b>/home/richardt/projects/apira-avatar/librealsense/common/model-views.cpp:21</b>:
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:549:22:</b> <font color="#CC0000"><b>error: </b></font>‘<b>const char* librealsense::get_string</b>’ redeclared as different kind of entity
  549 |     RS2_ENUM_HELPERS(<font color="#CC0000"><b>rs2_calib_target_type</b></font>, CALIB_TARGET)
      |                      <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:519:48:</b> <font color="#06989A"><b>note: </b></font>in definition of macro ‘<b>RS2_ENUM_HELPERS_CUSTOMIZED</b>’
  519 |     LRS_EXTENSION_API const char * get_string( <font color="#06989A"><b>TYPE</b></font> value );                                       \
      |                                                <font color="#06989A"><b>^~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:549:5:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro ‘<b>RS2_ENUM_HELPERS</b>’
  549 |     <font color="#06989A"><b>RS2_ENUM_HELPERS</b></font>(rs2_calib_target_type, CALIB_TARGET)
      |     <font color="#06989A"><b>^~~~~~~~~~~~~~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:519:36:</b> <font color="#06989A"><b>note: </b></font>previous declaration ‘<b>const char* librealsense::get_string(rs2_timestamp_domain)</b>’
  519 |     LRS_EXTENSION_API const char * <font color="#06989A"><b>get_string</b></font>( TYPE value );                                       \
      |                                    <font color="#06989A"><b>^~~~~~~~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:515:5:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro ‘<b>RS2_ENUM_HELPERS_CUSTOMIZED</b>’
  515 |     <font color="#06989A"><b>RS2_ENUM_HELPERS_CUSTOMIZED</b></font>( TYPE, 0, RS2_##PREFIX##_COUNT - 1 )
      |     <font color="#06989A"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:548:5:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro ‘<b>RS2_ENUM_HELPERS</b>’
  548 |     <font color="#06989A"><b>RS2_ENUM_HELPERS</b></font>(rs2_timestamp_domain, TIMESTAMP_DOMAIN)
      |     <font color="#06989A"><b>^~~~~~~~~~~~~~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:549:22:</b> <font color="#CC0000"><b>error: </b></font>‘<b>rs2_calib_target_type</b>’ was not declared in this scope; did you mean ‘<b>rs2_calibration_type</b>’?
  549 |     RS2_ENUM_HELPERS(<font color="#CC0000"><b>rs2_calib_target_type</b></font>, CALIB_TARGET)</pre>

There are more errors of this type about the ambiguous operator overload

<pre><b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:524:39:</b> <font color="#CC0000"><b>error: </b></font>invalid conversion from ‘<b>int</b>’ to ‘<b>rs2_timestamp_domain</b>’ [<font color="#CC0000"><b>-fpermissive</b></font>]
  524 |             return out &lt;&lt; get_string( <font color="#CC0000"><b>value</b></font> );                                                     \
      |                                       <font color="#CC0000"><b>^~~~~</b></font>
      |                                       <font color="#CC0000"><b>|</b></font>
      |                                       <font color="#CC0000"><b>int</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:515:5:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro ‘<b>RS2_ENUM_HELPERS_CUSTOMIZED</b>’
  515 |     <font color="#06989A"><b>RS2_ENUM_HELPERS_CUSTOMIZED</b></font>( TYPE, 0, RS2_##PREFIX##_COUNT - 1 )
      |     <font color="#06989A"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:549:5:</b> <font color="#06989A"><b>note: </b></font>in expansion of macro ‘<b>RS2_ENUM_HELPERS</b>’
  549 |     <font color="#06989A"><b>RS2_ENUM_HELPERS</b></font>(rs2_calib_target_type, CALIB_TARGET)
      |     <font color="#06989A"><b>^~~~~~~~~~~~~~~~</b></font>
<b>/home/richardt/projects/apira-avatar/librealsense/tools/realsense-viewer/../../src/types.h:526:24:</b> <font color="#CC0000"><b>error: </b></font>ambiguous overload for ‘<b>operator&lt;&lt;</b>’ (operand types are ‘<b>std::ostream</b>’ {aka ‘<b>std::basic_ostream&lt;char&gt;</b>’} and ‘<b>int</b>’)
  526 |             return <font color="#4E9A06">out</font> <font color="#CC0000"><b>&lt;&lt;</b></font> <font color="#3465A4">(int)value</font>;                                                              \
      |                    <font color="#4E9A06">~~~</font> <font color="#CC0000"><b>^~</b></font> <font color="#3465A4">~~~~~~~~~~</font>
      |                    <font color="#4E9A06">|</font>      <font color="#3465A4">|</font>
      |                    <font color="#4E9A06">|</font>      <font color="#3465A4">int</font>
      |                    <font color="#4E9A06">std::ostream {aka std::basic_ostream&lt;char&gt;}</font></pre>

MartyG-RealSense, 2021-06-24:
Hi @bontsai  I have not seen these particular errors before.  Most of the CMake build instruction that you are using is widely used, with the rarer part being the **-DCMAKE_INSTALL_PREFIX=/opt/realsense** statement to control where librealsense is built.

As a starting point in this investigation, I would therefore suggest removing this command from the CMake build instruction to see whether it makes a positive difference.

bontsai, 2021-06-24:
Hi @MartyG-RealSense, so I haven't tried to run it, but to clarify the DCMAKE_INSTALL_PREFIX=/opt/realsense actually only controls where the libraries are installed from the make install.  Not where the librealsense is built.  It is still built in the build directory where you run the cmake .. from

I will inform you if it works without.  I'm still guessing this is still a problem and a bit miffed that nobody has actually run into this problem for three different versions.  As I mentioned it built for 2.42 and because of new hardware I needed to upgrade as that old version didn't detect the camera.  and started with 2.47 and went one by one until 2.43 worked and I was able to use it with the new camera.

MartyG-RealSense, 2021-06-25:
Okay, thanks very much @bontsai - I look forward to hearing whether the build works without using **DCMAKE_INSTALL_PREFIX**

MartyG-RealSense, 2021-07-01:
Hi @bontsai  Do you require further assistance with this case, please?  Thanks!

bontsai, 2021-07-02:
Hi @MartyG-RealSense - I actually tried what you suggested and it seems to work.  Ironically I got the recent v2.48 and was able to use the one with the CMAKE flag for the install directory to work.  I did not bother to try with the other versions, since I'll probably just go with the latest.  But I still do think there is some issue with possibly the CMAKE flag for install directory which was introduced in v2.44

MartyG-RealSense, 2021-07-03:
Thanks very much @bontsai for the confirmation that the CMake flag **DCMAKE_INSTALL_PREFIX** works for you in 2.48.0.

closed 2021-07-15 by MartyG-RealSense