realsenseai/librealsense #9282: Questions about the way D435 avoids obstacles

ju-mingyue, 2021-06-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |                                  D435              | 
| Firmware Version                   |05.12.13.50 | 
| Operating System & Version |  Linux (Ubuntu18.04.5)  | 
| Kernel Version (Linux Only)    |  4.9.201                                         | 
| Platform                                 | NVIDIA JetsonNano B01 |
| SDK Version                            |  2.41.0                         | 
| Language                            | ROS packages }                          | 
| Segment			|  Robot                       | 

### Issue Description
Hello, I need to use the obstacle avoidance function in the process of using D435. At present, there are two ways I have inquired:
1. Use depthimage_to_laserscan to convert the depth information into a lidar signal, but the current problem is that there is also a lidar on my robot. Now these two topics are scan, so there is a conflict. I don't know how to solve it.

2. I want to know whether the two lidar signals can be fused, and what configuration is needed to be fused. Is there any relevant information or code?

3. Using plotcloud2 point cloud information, I don't understand how to do this at present. Although the point cloud image can be seen on the map now, it does not have the effect of avoiding obstacles. And does this point cloud information need to be passed to AMCL? If so, how does it need to be delivered? So I hope someone can help me.

ju-mingyue, 2021-06-24:
@MartyG-RealSense Thank you very much for your reply，The solution to the previous question（Questions about the use of D435 #9272
） does not answer the problem I am facing now, so I hope you will give me some other answers

MartyG-RealSense, 2021-06-24:
Hi @ju-mingyue  I have researched your case and was just putting together a reply.  There is a ROS package called **ira_laser_tools** that can merge multiple laser scans together.

https://github.com/iralabdisco/ira_laser_tools

Information about its use can be found in a paper at the link below.

https://arxiv.org/abs/1411.1086

closed 2021-07-02 by ju-mingyue