realsenseai/librealsense #9288: How to save color image and depth map?

JZhu1208, 2021-06-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | v2.45.0 | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            |   v2.45.0  }                          | 
| Language                            |  {C++ matlab }                          | 
| Segment			|                            | 

### Hello,

I am a beginner to Intel RealSense L515, I connect the camera to the computer, and I can view the color image and depth map in the RealSense viewer. I want to know how to save them as .jpg or .png or other formats?

In addition, there is no rs-convert.exe in my C:\Users\ZJ\Downloads\librealsense-2.45.0\librealsense-2.45.0\tools\convert, where can I get it? Thank you

Sincerely

MartyG-RealSense, 2021-06-26:
Hi @JZhu1208  In the **2D** mode of the RealSense Viewer, the panel of each stream type that is currently enabled has a row of icons at the top of it.  They include options to **Pause** the stream and also to export a **Snapshot** in PNG image format.

[image]

You can use the Pause option to pause the live stream (or so so with the space bar of the keyboard) and then use the camera-shaped Snapshot option to export the current frame.

When saving a Snapshot from the Depth or Infrared stream, additional files are automatically created in **.raw** format (a 'depth matrix' image format that preserves depth information) and **.csv** format (a textual file that can be opened in database and spreadsheet software). 

[image]

If you require more than one frame then you can use the **Record** option at the top of the Viewer's options side-panel to record a **bag** file.  This is like a video recording of camera data that captures all of the streams that are currently enabled when the recording begins.  Left-clicking on the Record option starts the recording and left-clicking again on the option ends the recording.

[image]

When recording is enabled, a blinking red light appears in the top corner of the stream panel appears, and disappears when the recording is closed.

[image]

Once a bag file recording has been created then you can replay it by drag-and-dropping the bag file into the center panel of the Viewer, or by going to the **Add Source** option at the top of the side-panel and selecting **Load Recorded Sequence** from its drop-down menu.

[image]

You can convert a bag file into a range of different formats using the RealSense SDK's **rs-convert** tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

If you have installed the RealSense SDK on your Windows computer, a pre-made executable version of rs-convert that can be launched immediately is located in the SDK's **Tools** folder.  rs-convert exe should be launched in the **Command Prompt** from the Tools folder.  Instructions for doing so are provided below.

1.  Go to the Tools folder.  You can locate this easily on Windows by right-clicking on the RealSense Viewer program launch icon that the SDK installer will have placed on the Windows desktop and selecting the **Open file location** menu option.

2.  In the address bar at the top of the Tools folder window, left-click on the end of the address text-box to convert it to the Windows-format address **C://Program Files (x86)/Intel RealSense SDK 2.0/Tools**

3.  Copy the contents of the address box with **CTRL-C** and go to the Command Prompt window.  You can find the Command Prompt by typing **command** into the search box at the bottom of the Windows screen, next to the corner button where the computer is shut down.

4.  Type cd followed by a space and use **CTRL-V** to paste the address into the Command Prompt window and press return. 
 The directory will change to the Tools folder.

5.  Input the rs-convert conversion command.   See the documentation for the rs-convert tool for information on structuring a bag conversion command.

JZhu1208, 2021-06-28:
Hi,

Thank you very much for your patient answers. Following your instructions, I have successfully saved the color image and depth map.

Encountered a new problem, is it possible to save the depth map as a file in .pfm format?

MartyG-RealSense, 2021-06-28:
It's excellent news @JZhu1208 that you were able to achieve saving the color image and depth map.

Regarding saving the depth map as a .pfm file, I do not know of a way to export directly to pfm using a librealsense script. The SDK can export in the **.raw** format though (an image format that preserves depth detail), and there is a tool called **Filestar** that can perform raw to pfm conversion.

https://filestar.com/skills/raw/convert-raw-to-pfm

JZhu1208, 2021-06-28:
Hello MartyG, thanks for the quick response.

I successfully used Filestar to convert raw files to pfm files. But since I don't have a .pfm display software, I use Filestar to convert the raw file to png to verify the correctness.

At this time, some problems appeared. The first image is capture by RealSense Viewer, it shows the correct depth map. The second image is converted by Filestar, it seems to have some errors (I am not sure if it should be called an error). On this question, do you have any experience?

[image]

[image]

MartyG-RealSense, 2021-06-28:
I have no reference for what a 'correct' pfm depth image derived from RealSense data should look like in order to do a comparison, unfortunately.  There is an interesting non-RealSense discussion in the link below though about another case in which a pfm image did not meet expectations of what it should look like.

https://github.com/roboception/cvkit/issues/5

JZhu1208, 2021-06-29:
I read the non-realsense discussion. But I think it is different from the problem I encountered. What I want to understand now is that the depth map （.png）captured by RealSense Viewer is different from the depth map converted from .raw files to .png files by Filestar. The first image is capture by RealSense Viewer, it shows the correct depth map. The second image is converted by Filestar.

[image]

[image]

MartyG-RealSense, 2021-06-29:
Depth data that is exported as a PNG from the RealSense SDK loses most of its depth information, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/3640

A .raw image's structure preserves the depth information though:

https://github.com/IntelRealSense/librealsense/issues/1485

MartyG-RealSense, 2021-07-05:
Hi @JZhu1208  Do you require further assistance with this case, please?  Thanks!

JZhu1208, 2021-07-06:
Thank you for your help. Regarding this question, I don’t think I need help anymore. You can close this issue. Thank you!

MartyG-RealSense, 2021-07-06:
You are very welcome @JZhu1208 - thanks for the update!  As suggested, I will close the issue as you are satisifed with the outcome.

closed 2021-07-06 by MartyG-RealSense