realsenseai/librealsense #9289: Failed to register the server's public key

JJONSOO, 2021-06-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    | 4.9.201                                       | 
| Platform                                 |  NVIDIA Jetson
| SDK Version                            |  2.47.0                        | 
| Language                            |  C++                        | 
| Segment			|  Devolopment                         | 

### Issue Description
[https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)
When installing packages from the following website, I have an <keyserver receive failed: No name> error when trying to register the server's public key.
Please let me know if you can help

MartyG-RealSense, 2021-06-25:
Hi @JJONSOO   As a first step in investigating your case, I would recommend checking whether the problem is occurring because of a **firewall**, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/1276

JJONSOO, 2021-06-28:
Hi, I finished everything on the site you pointed out and tried to execute the example file, but I have an error 
api version mismatch : librealsense.so was compiled with API version 2.41.0 but the application was compiled with 2.42.0 make sure correct version of the library is installed 
What should I do?

MartyG-RealSense, 2021-06-28:
The error suggests that the example program was created in a version of librealsense that is newer than the one that you have installed from packages.

There was a recent case in which a RealSense user was installing librealsense packages for their Jetson but it was only librealsense **2.41.0** instead of the latest **2.47.0** version.  **Doronhi** the RealSense ROS wrapper developer commented on this in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1938#issuecomment-867379211

The RealSense user in that case instead built librealsense for their Jetson from source code using the instructions in doronhi's comment.

hanzheteng, 2021-07-02:
Hi @MartyG-RealSense,

I encountered the same issue. It seems like the instructions in section 5. Register the server's public key in [installation_jetson](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) is the same as the one in [distribution_linux](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).

The bottleneck is on this command: `sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE`  It will lead to the error `gpg: keyserver receive failed: No name` on my Jetson Xavier NX device. 

PS. I used to successfully installed everything on a Jetson AGX Xavier device following the same instructions as in distribution_linux about half a year ago.

MartyG-RealSense, 2021-07-03:
Hi @hanzheteng   I recommend using the Jetson installation page's package instructions for installing packages on Jetson (section **4.  Install with Debian Packages**) as the Debian packages on the distribution_linux page are designed for use with x86 / x64 computers. 

My research of your question found though that problems with using **sudo apt-key adv --keyserver keys.gnupg.net --recv key** have been reported recently at the end of June 2021 at the link below, which offers advice for resolving the problem.

https://stackoverflow.com/questions/66217436/gpg-keyserver-receive-failed-no-name/68132500#68132500

hanzheteng, 2021-07-03:
Hi @MartyG-RealSense ,

As I mentioned earlier, this error also occurred on amd64 machines, so it is not related to Jetson platform. (Also, there is no actual instruction under **4. Install with Debian Packages** if you take a look.)

The main issue here is that the key server `keys.gnupg.net` is not functioning anymore. Someone reported this recent issue [here](https://stackoverflow.com/questions/66217436/gpg-keyserver-receive-failed-no-name).

The solution is to replace the key server `keys.gnupg.net` by `keyserver.ubuntu.com` or other functioning ones. In both instructions ([installation_jetson](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) and [distribution_linux](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)), use the following command to register the server's public key should fix this problem. 

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE`

Please double check and update the instructions accordingly.

MartyG-RealSense, 2021-07-03:
https://github.com/IntelRealSense/librealsense/issues/9315#issuecomment-871250227 was not only related to Jetson, so it was a valid suggestion for a problem that could occur on both x86/x64 and the Arm-based Jetson.

The link about **keys.gnupg.net** was my original message to you, which I edited out because I believed that https://github.com/IntelRealSense/librealsense/issues/9315#issuecomment-871250227 was worth looking at first as a more likely possibility.  So it's great to hear that the original suggestion worked for you after all.

I will highlight the above information to Intel.  Thanks very much for the update!

hanzheteng, 2021-07-03:
Thank you for the time and effort for helping out. The original message did help. Please convey the message to Intel about the possibly deprecated key server then.

MartyG-RealSense, 2021-07-06:
Hi @hanzheteng  I have received a response from Intel about this issue.  They said that the official instruction pages were updated several months ago with the alternate '||' keyserver address.

**distribution_linux.md**
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE **|| sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key** F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE

**Jetson**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE **|| sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key** F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE

hanzheteng, 2021-07-07:
Thank you for the clarification. I just verified on a clean Ubuntu 18 virtual machine that the second part also works. 

The behavior of the first part varies on different machines. On some machines it returns immediately, but on some other machines it can take 10s to return the No Name error. The first time I ran this line of command, the first part was not responding and interrupted by me, so the second part was not actually being executed.

MartyG-RealSense, 2021-07-07:
Thanks very much @hanzheteng for your test feedback!

closed 2021-07-19 by MartyG-RealSense