realsenseai/librealsense #9290: D435 camera not working on nvgstcapture-1.0

Edu4444, 2021-06-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.201-tegra                                         | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  { legacy / 2.47.0 }                          | 
| Language                            |                          | 
| Segment			| Deep learning | 

### Issue Description
Hello. I'm trying to use D435 camera on jetson-infernce detectnet models.
I have build from source using native backend.
D435 camera is detected when using` $ ls /dev/video*` but running the camera using detecnet or `nvgstcapture-1.0 --camsrc=0 --cap-dev-node=2` does not show any images.

```
mero@Jetson-HC02:~$ ls /dev/video*
/dev/video0  /dev/video1  /dev/video2
mero@Jetson-HC02:~$ nvgstcapture-1.0 --camsrc=0 --cap-dev-node=2
Encoder null, cannot set bitrate!
Encoder Profile = High
Runtime USB Camera Commands:

  Help : 'h'
  Quit : 'q'
  Set Capture Mode:
      mo:<val>
          (1): image
          (2): video
  Get Capture Mode:
      gmo
  Capture: enter 'j' OR
           followed by a timer (e.g., jx5000, capture after 5 seconds) OR
           followed by multishot count (e.g., j:6, capture 6 images)
           timer/multihot values are optional, capture defaults to single shot with timer=0s
  Start Recording : enter '1'
  Stop Recording  : enter '0'
  Set Preview Resolution:
      pcr:<val> e.g., pcr:2
          (0) : 176x144
          (1) : 320x240
          (2) : 640x480
          (3) : 1280x720
          (4) : 1920x1080
  NOTE: Preview/Encode resolution will be same as Capture resolution for USB-Camera
  Get Preview Resolution:
      gpcr
  Get Image Capture Resolution:
      gicr
  Get Video Capture Resolution:
      gvcr
  Set Capture Device Node:
      cdn:<val> e.g., cdn:0
          (0): /dev/video0
          (1): /dev/video1
          (2): /dev/video2
  Get Capture Device Node:
      gcdn

Runtime encoder configuration options:

  Set Encoding Bit-rate(in bytes):
      br:<val> e.g., br:4000000
  Get Encoding Bit-rate(in bytes):
      gbr
  Set Encoding Profile(only for H.264):
      ep:<val> e.g., ep:1
          (0): Baseline
          (1): Main
          (2): High
  Get Encoding Profile(only for H.264):
      gep
  Force IDR Frame on video Encoder(only for H.264):
      Enter 'f' 

** Message: 11:30:51.955: <main:4670> iterating capture loop ....
ERROR on bus: by /GstPipeline:capture_native_pipeline: Output window was closed
debug info:
/dvs/git/dirty/git-master_linux/tests-multimedia/nvgstapps/nvgstcapture-1.0/nvgstcapture.c(4112): nvgst_handle_xevents (): /GstPipeline:capture_native_pipeline
** Message: 11:30:53.965: <main:4680> Capture completed
** Message: 11:30:53.967: <main:4729> Camera application will now exit

```

MartyG-RealSense, 2021-06-25:
Hi @Edu4444  In a discussion in the link below, a RealSense user was able to access a D435 with detectnet-camera by grabbing the RGB V4L2 stream from /dev/video2 when running the command below:

**./detectnet-camera --camera /dev/video2**

https://github.com/IntelRealSense/librealsense/issues/5994

Is the above link helpful in solving your problem, please?

MartyG-RealSense, 2021-07-02:
Hi @Edu4444  Do you require further assistance with this case, please?  Thanks!

Edu4444, 2021-07-02:
> Hi @Edu4444 Do you require further assistance with this case, please? Thanks!

Thanks for the reply.
I have been using `./detectnet /dev/video2` to run the model.
I have to wait a couple of days until I have access to the jetson nano again.
I will post a reply as soon as I test the detectnet-camera program.

MartyG-RealSense, 2021-07-02:
Okay, that's no problem at all   Thanks for the update!

Edu4444, 2021-07-06:
Hello again.
I have tried `./detectnet-camera --camera /dev/video2` command, but it has the same output as `./detectnet /dev/video2`.

```
mero@Jetson-HC02:~/jetson-inference/build/aarch64/bin$ ./detectnet /dev/video2
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video2
[gstreamer] gstCamera -- found v4l2 device: Intel(R) RealSense(TM) Depth Ca
[gstreamer] v4l2-proplist, device.path=(string)/dev/video2, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Intel\(R\)\ RealSense\(TM\)\ Depth\ Ca", v4l2.device.bus_info=(string)usb-70090000.xusb-1.1, v4l2.device.version=(uint)264649, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 9 caps for v4l2 device /dev/video2
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 15/1, 6/1 };
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 15/1, 6/1 };
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)960, height=(int)540, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 15/1, 6/1 };
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)848, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 15/1, 6/1 };
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 15/1, 6/1 };
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 15/1, 6/1 };
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)424, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 15/1, 6/1 };
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 6/1 };
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 30/1, 6/1 };
[gstreamer] gstCamera -- selected device profile:  codec=raw format=yuyv width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video2 ! video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720 ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video2
[video]  created gstCamera from v4l2:///dev/video2
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video2
     - protocol:  v4l2
     - location:  /dev/video2
     - port:      2
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1920
  -- height:     1080
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[... 85 lines cut ...]
[TRT]    W = 7  H = 100  C = 1
[TRT]    detectNet -- maximum bounding boxes:  100
[TRT]    detectNet -- loaded 91 class info entries
[TRT]    detectNet -- number of object classes:  91
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> v4l2src0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> v4l2src0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> v4l2src0
[gstreamer] gstreamer message stream-start ==> pipeline0
detectnet:  failed to capture video frame
detectnet:  failed to capture video frame
detectnet:  failed to capture video frame

```

For some reason the RGB V4L2 stream is not accessible. I only can visualice RGB frames in RealSense Viewer.

MartyG-RealSense, 2021-07-06:
The link below advises that raw data from a RealSense camera can be captured by Linux tools if the SDK is built with the V4L2 backend using the CMake build flag **-DFORCE_LIBUVC=OFF**

https://github.com/IntelRealSense/librealsense/issues/6841#issuecomment-660859774

Edu4444, 2021-07-07:
Thanks for the support.
After uninstaling and cleaning the installation I have built again the code using the following command:
`mero@Jetson-HC02:~/librealsense/build$ cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true -DFORCE_LIBUVC=OFF && make -j$(($(nproc)-1)) && sudo make install`
 Unfortunately the issue is still happening.

MartyG-RealSense, 2021-07-07:
I researched the issue further and found a note in the **Nvidia Jetson Linux Developer Notes** document that may be relevant.

************

nvgstcapture-1.0 has an image encoding issue using the nvjpegenc plugin for default YUY2 video capture format for a USB camera.  To work around the issue, use the OSS **jpegenc** plugin instead of nvgstcapture1.0 for image encoding with a USB camera. 

To do this, add the command line switch **--image-enc=0** to the nvgstcapture-1.0 capture command, for
example:

$ nvgstcapture-1.0 --camsrc=0 --image-enc=0

Edu4444, 2021-07-08:
I have tested using `nvgstcapture-1.0 --camsrc=0 --image-enc=0 --cap-dev-node=2` but generated window is not receiving any frame from the pipeline. I had to stop the pipeline using ctrl + c.

```
** Message: 12:45:06.957: <main:4670> iterating capture loop ....
^C** Message: 12:45:21.512: <_intr_handler:4261> User Interrupted.. 

Terminating the camera pipeline ...
** Message: 12:45:21.571: <main:4680> Capture completed
** Message: 12:45:21.572: <main:4729> Camera application will now exit
```

The D435 camera is working as expected on a jetson TX2.
I have also tested nvgstcapture with an old USB camera successfully in the same jetson nano where D435 pipeline is not working.

MartyG-RealSense, 2021-07-08:
And you are not able to obtain an image if you use a different **/dev/video** channel by using '0' or '1' as the --cap-dev-node value instead of '2'?

Edu4444, 2021-07-08:
The other nodes don't work. With node 2 the program generates a window and it keeps waiting frames.

MartyG-RealSense, 2021-07-08:
I believe that all the relevant reference sources for nvgstcapture-1.0 have now been explored, unfortunately.

An alternative approach may be to use a different video tool such as **Gstreamer** instead of nvgstcapture-1.0.  I translated an article into English about using a D435 and Gstreamer with detectnet and have attached the PDF to this message.

[DetectNet-D435-Gstreamer.pdf](https://github.com/IntelRealSense/librealsense/files/6786603/DetectNet-D435-Gstreamer.pdf)

Edu4444, 2021-07-09:
Detectnet scripts use Gstreamer by default. As you can see in the following comment, detectnet fails capturing video frames using Gstreamer.
https://github.com/IntelRealSense/librealsense/issues/9290#issuecomment-874500413

I think the issue may be related to this specific jetpack version on jetson nano or the building process. I have rebuilt the code several times using this guide with native backend and applying `./scripts/patch-realsense-ubuntu-L4T.sh`.
https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation#section-building-from-source-using-native-backend

MartyG-RealSense, 2021-07-09:
I note from the information at the top of this case that you are using JetPack **4.5.1**.  Support for JetPack 4.5 (32.5) was added in SDK version **2.43.0** but not JetPack 4.5.1 at the time of writing this.

MartyG-RealSense, 2021-07-13:
Looking at the problem in regard to nvgstcapture-1.0 was a dead-end in regards to research leads.  So I researched it as a detectnet problem instead.  

I located a case on the Nvidia forums regarding a blank screen when using a D435 with detectnet.  Nvidia forum staff provide a range of suggestions in that discussion for possible ways to address the problem.

https://forums.developer.nvidia.com/t/issue-in-connecting-camera/71177

Edu4444, 2021-07-14:
I have the camera finally working!
I have installed Jetpack 4.3 (L4T 32.3.1) and I have built the project following the instructions from JetsonHacks:
https://www.jetsonhacks.com/2019/12/22/install-realsense-camera-in-5-minutes-jetson-nano

It seems recent SDK versions does not work properly with Jetpack 4.4.1 on the Jetson Nano.
Is there any plans for adding support for Jetpack 4.5.1?

closed 2021-07-14 by Edu4444

MartyG-RealSense, 2021-07-14:
Excellent news @Edu4444 that you achieved a solution!  Thanks very much for sharing the details of it.

It is likely that JetPack support will be updated further at a future date based on past JetPack support, but I do not have any information about time schedules or which JetPacks would be added to the list of supported versions.

Thanks again!