realsenseai/librealsense #9291: WEB UI to control camera parameters - searching for suggestions / advices

devaale, 2021-06-25:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |  Win10  | 
| Platform                                 | PC |
| Language                               |  Python | 

Hello, I'm trying to build WEB UI on python - to control camera parameters while the camera is reading frames (for example changing post-processing steps, their params, no streaming to WEB required) - maybe anyone built something similar and could advise on technologies to use?  - Would Flask, Vue and Sockets suffice ? (everything gonna be working locally) Any suggestions would be helpful as I'm researching at the moment how I could go about this.

MartyG-RealSense, 2021-06-25:
Hi @devaale  As a starting point for your research, the three links below demonstrate approaches that other RealSense users have taken with Python projects that make use of Flask.

https://github.com/IntelRealSense/librealsense/issues/6475#issuecomment-635990864
https://freesoft.dev/program/155363707
https://github.com/WEMarsSoftware/Cam-Server

devaale, 2021-06-27:
@MartyG-RealSense Appreciate the help ! Gonna look into all of em and cherry-pick pieces I need. THANK YOU !

MartyG-RealSense, 2021-06-27:
You are very welcome @devaale - thanks for the update, and good luck!

MartyG-RealSense, 2021-07-09:
Case closed due to no further comments received.

closed 2021-07-09 by MartyG-RealSense

devaale, 2021-09-17:
@MartyG-RealSense Sorry for the late response! Project ended up expanding :D Appreciate your help once more! I will post results sometime later when I'm fully done with the system for people in the future to use <3

MartyG-RealSense, 2021-09-18:
No problem at all, @devaale - thanks very much for the update and your willingness to share your knowledge. :)