realsenseai/librealsense #9292: DKMS installation problem with kernel 5.8

kevindehecker, 2021-06-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0.                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.47                          | 

### Issue Description

I'm having trouble reading the exposure from my D455 (which I believe is part of the advanced features). During re-installation of the lbrealsense I encountered the following:

```
sudo apt install librealsense2-dkms 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  librealsense2-dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 12.3 MB of archives.
After this operation, 65.8 MB of additional disk space will be used.
Get:1 https://librealsense.intel.com/Debian/apt-repo bionic/main amd64 librealsense2-dkms all 1.3.18-
0ubuntu1 [12.3 MB]
Fetched 12.3 MB in 2s (6,562 kB/s)             
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 147656 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.18-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.18-0ubuntu1) ...
Setting up librealsense2-dkms (1.3.18-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.18 DKMS files...
Building for 5.8.0-41-generic 5.8.0-59-generic
Building initial module for 5.8.0-41-generic
Error!  The /var/lib/dkms/librealsense2-dkms/1.3.18/5.8.0-41-generic/x86_64/dkms.conf for module libr
ealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.
Building initial module for 5.8.0-59-generic
Error!  The /var/lib/dkms/librealsense2-dkms/1.3.18/5.8.0-59-generic/x86_64/dkms.conf for module libr
ealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.18: added
mod: videodev                   
mod: uvcvideo                    version: 1.1.1
mod: hid_sensor_gyro_3d         
mod: hid_sensor_accel_3d
```

Not sure if related to the exposure problem, but at least it's an error Intel can want to fix?

MartyG-RealSense, 2021-06-25:
Hi @kevindehecker  This particular error with Kernel 5.8. has arisen before in a case in March 2021 (linked to below).

https://github.com/IntelRealSense/librealsense/issues/8283#issuecomment-808908659

It was advised in that case that metadata support should be enabled by patching kernel 5.8.  

https://github.com/IntelRealSense/librealsense/issues/8283#issuecomment-825587976

Kernel patch support for 5.8 was added in SDK 2.45.0.

[image]

DKMS package support for 5.8 was, of course, added in SDK **2.47.0**.

kevindehecker, 2021-06-25:
Aha. OK 2 things.

1.
I ran the `./scripts/patch-realsense-ubuntu-lts.sh` patch successfully, which coincidentally did solve my initial problem of not having access to the exposure data. Nice, thanks for pointing that out! 

2.
> DKMS package support for 5.8 was, of course, added in SDK 2.47.0.

Just as a sanity check I reran the DKMS installation and the error persists... (although it now has no known side-effects for me at least). Weird...?

MartyG-RealSense, 2021-06-26:
Thanks very much @kevindehecker for the update - it's excellent to hear that you were successful.  Please do feel free to report if you experience any further problems with the DKMS installation.

ev-mp, 2021-06-29:
@kevindehecker hello,
That the error is due to a mismatch between the Ubuntu version you're running  (Focal ) and the repo it is currently mapped to : 
>Get:1 https://librealsense.intel.com/Debian/apt-repo **bionic**/main amd64 librealsense2-dkms all 1.3.18-

For Ubuntu Bionic the DKMS package supports up to kernel 5.4 according to Ubuntu roadmap).
Kernel 5.8 is applicable for Focal only at this stage.
Update the APT sources with the [proper PPA](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) to resolve the inconsistency

kevindehecker, 2021-07-03:
Yes, that was it! It used to be that in the docs there was a separate mention for Ubuntu Bionic/Focal/... etc, so I scanned for that when we started the upgrade to Focal last week but I didn't see. So I just implicitly assumed the old repo was now universal (I script the install, realsense is just a small part of it). 

Oops :)

Thanks!

closed 2021-07-03 by kevindehecker