realsenseai/librealsense #9293: Finding updated intrinsic values when using On-Chip Calibration

OiCata44, 2021-06-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |  Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  5.4.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0                          | 
| Language                            |  python                          | 
 

### Issue Description
Hi there,
I've been utilizing the On-Chip calibration feature and its been working great. I keep getting good feedback from the health check (<.25). However I'm interested in exactly what fields are changed within the calibration table, but when I check the table within realsense viewer it doesn't seem to update. So I switched to the **depth_auto_calibration_example.py** script in the SDK. I'm still getting a good health check but the array outputted by the **get_calibration_table()** function still doesn't change between calibrations. Am I missing a change that affects the health check but isn't stored in the calibration table? Is there a simple way to find the updated intrinsic fields and their new values?
Thanks.

MartyG-RealSense, 2021-06-26:
Hi @OiCata44  Once you have performed a calibration and are satisfied with the result, are you saving it to the calibration table stored inside the camera hardware, please?  In the Viewer's on-chip calibration, that would be done with the **Apply Now** option.

[image]

You can check the calibration information in the Viewer with the **Calibration Data** option of the 'More' menu.  Is this the method that you used to check the table in the Viewer?

[image]

[image]

My understanding is that the above table represents the calibration table values that are updated by an On-Chip depth calibration.

The SDK's **Dynamic Calibration** tool - an alternative to the On-Chip calibration method - provides a simlar list of calibration parameters in its documentation, with the addition of parameters for the rotation and translation of the RGB sensor (as the Dyamic Calibration tool can calibrate both depth and RGB sensors).

[image]

OiCata44, 2021-06-28:
@MartyG-RealSense This was super helpful! Looks like the calibration changes were so small it wasn't appearing within the default three decimal places on the Realsense Viewer GUI. After comparing the differences in saved calibration tables through CLI all the pieces fit. I can easily see which Intrinsic values were updated now, thanks for your help! 
[image]

MartyG-RealSense, 2021-06-28:
Excellent news @OiCata44 that you were able to confirm that the intrinsics were changing after calibration.  Thanks very much for the update!

closed 2021-07-10 by MartyG-RealSense