realsenseai/librealsense #9295: Suggestion: include zdepth lossless depth compression

roelofvandijk, 2021-06-26:
To decrease the bandwidth of streaming depth data, it could be interesting to add a post-proccesing option to losslessly compress the depth (in a different manner than described in the [colorization/compression whitepaper](https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras)).

An example implementation with a permissive licence can be found here: https://github.com/catid/Zdepth

MartyG-RealSense, 2021-06-26:
Hi @roelofvandijk  A compression format called **Z16H** was added to the SDK in version **2.32.1**, along with a processing block **(rs2_create_huffman_depth_decompress_block**) for using Huffman decompression to decompress the Z16H format.

https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2321
https://intelrealsense.github.io/librealsense/doxygen/rs__processing_8h.html#ae2a413b273e6134503701e4fa51b3793

I am not aware of any applications or scripting references that make use of the Z16H format though, and in late 2020 the RealSense ROS wrapper added mechanisms to avoid using it due to it being 'currently unsupported'.

MartyG-RealSense, 2021-07-03:
Hi @roelofvandijk  Do you require further assistance with this case, please?  Thanks!

roelofvandijk, 2021-07-04:
Hello @MartyG-RealSense, that is great news, thank you.
It would be nice to know whether this compression is available or just for internal use, I cannot find it in the docs/API/source code. If it is for external use, a benchmark would be nice.

MartyG-RealSense, 2021-07-05:
I researched the issue further and located a C++ script shared by a RealSense user that uses Z16H as the depth format when **recording** to a bag file.

https://github.com/IntelRealSense/librealsense/issues/8726

[image]

roelofvandijk, 2021-07-05:
Hello @MartyG-RealSense, thank you, I will try that!

closed 2021-07-05 by roelofvandijk