realsenseai/librealsense #9302: Access denied AWS container cmake connectin check file

ankistr, 2021-06-28:
I tried to build the c++ project with the cmake_lists.txt but the build fails. It fails due to the problem, that the required AWS container sends "Access denied". (AWS container http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/connectivity_check)

Below cmake_lists file and connectivity check file. 

Does anyone has an idea how to solve this problem ? 

```
cmake_lists.txt
#  minimum required cmake version: 3.1.0
cmake_minimum_required(VERSION 3.2.0)
project(librealsense2 LANGUAGES CXX C)

include(CMake/lrs_options.cmake)
include(CMake/connectivity_check.cmake)

#Deprecation message, should be removed in future releases
if(${FORCE_LIBUVC} OR ${FORCE_WINUSB_UVC} OR ${ANDROID_USB_HOST_UVC})
    MESSAGE(DEPRECATION "FORCE_LIBUVC, FORCE_WINUSB_UVC and ANDROID_USB_HOST_UVC are deprecated, use FORCE_RSUSB_BACKEND instead")
    set(FORCE_RSUSB_BACKEND ON)
endif()

# Checking Internet connection, as TM2 needs to download the FW from amazon cloud
if(BUILD_WITH_TM2 AND NOT INTERNET_CONNECTION)
    message(WARNING "No internet connection, disabling BUILD_WITH_TM2")
    set(BUILD_WITH_TM2 OFF)
endif()

# Checking Internet connection, as DEPTH CAM needs to download the FW from amazon cloud
if(IMPORT_DEPTH_CAM_FW AND NOT INTERNET_CONNECTION)
    message(WARNING "No internet connection, disabling IMPORT_DEPTH_CAM_FW")
    set(IMPORT_DEPTH_CAM_FW OFF)
endif()

list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_LIST_DIR}/CMake)

# include librealsense general configuration
include(CMake/global_config.cmake)

config_cxx_flags()

# include os specific macros
# macro definition located at "CMake/global_config.cmake"
include(CMake/include_os.cmake)

# set os specific configuration flags
# macro definition located at "CMake/<OS>_config.cmake"
os_set_flags()

# set global configuration flags
# macro definition located at "CMake/global_config.cmake"
global_set_flags()

if(BUILD_SHARED_LIBS)
    add_library(${LRS_TARGET} SHARED "")
else()
    add_library(${LRS_TARGET} STATIC "")
endif()

# set library version
set_target_properties(${LRS_TARGET} PROPERTIES VERSION ${REALSENSE_VERSION_STRING} SOVERSION "${REALSENSE_VERSION_MAJOR}.${REALSENSE_VERSION_MINOR}")

include(include/CMakeLists.txt)
include(src/CMakeLists.txt)
include(third-party/CMakeLists.txt)

# configure the project according to OS specific requirments
# macro definition located at "CMake/<OS>_config.cmake"
os_target_config()

# global project configuration
# macro definition located at "CMake/global_config.cmake"
global_target_config()

include(CMake/install_config.cmake)

add_subdirectory(wrappers)

if (BUILD_EXAMPLES AND BUILD_GLSL_EXTENSIONS)
    find_package(glfw3 3.3 QUIET)
    if(NOT TARGET glfw)
        message(STATUS "GLFW 3.3 not found; using internal version")
        set(GLFW_INSTALL ON CACHE BOOL "" FORCE)
        add_subdirectory(third-party/glfw)
    endif()
    add_subdirectory(src/gl)
endif()

# User specific paths for draco dynamic lib
# In the target CMakeLists.txt add:
#       target_link_libraries(<target> ${DRACO_LIBRARY})
#       target_include_directories(<target> ${DRACO_INCLUDE_PATH})
if (USE_EXTERNAL_DRACO)
    message(STATUS "Using external draco lib")
    set(DRACO_INCLUDE_PATH "C:/Program Files/draco/include")
    set(DRACO_DLL_PATH "C:/Program Files/draco/lib")
    find_library(DRACO_LIBRARY draco ${DRACO_DLL_PATH})
else(USE_EXTERNAL_DRACO)
    set(DRACO_INCLUDE_PATH ${CMAKE_SOURCE_DIR}/third-party/draco/src)
    add_subdirectory(third-party/draco)
endif () #USE_EXTERNAL_DRACO

if(BUILD_EXAMPLES)
    add_subdirectory(examples)
    add_subdirectory(tools)
endif()

# Add experiments to build
if (BUILD_EXPERIMENTS)
    add_subdirectory(holotarting-src)
endif ()

if(BUILD_UNIT_TESTS)
    add_subdirectory(unit-tests)
endif()

if(BUILD_NETWORK_DEVICE)
    add_subdirectory(src/ethernet)
    add_subdirectory(src/compression)
endif()

if(BUILD_WITH_TM2)
    add_tm2()
endif()

if(IMPORT_DEPTH_CAM_FW)
    add_subdirectory(common/fw)
endif()

include(CMake/embedd_udev_rules.cmake)
```

``
CMake/connectivity_check.cmake

# License: Apache 2.0. See LICENSE file in root directory.
# Copyright(c) 2019 Intel Corporation. All Rights Reserved.
message(STATUS "Checking internet connection...")
file(DOWNLOAD "http://realsense-hw-public.s3-eu-west-1.amazonaws.com/Releases/connectivity_check" "${CMAKE_CURRENT_SOURCE_DIR}/connectivity_check" SHOW_PROGRESS TIMEOUT 5 STATUS status)
list (FIND status "\"No error\"" _index)
if (${_index} EQUAL -1)
    message(STATUS "Failed to identify Internet connection")
    set(INTERNET_CONNECTION OFF)
else()
    message(STATUS "Internet connection identified")
    set(INTERNET_CONNECTION ON)
endif()

[image]

ev-mp, 2021-06-28:
@ankistr , starting with v2.42 the SDK [has switched](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features-3) to HTTPS URLs in Cmake builds.
The old URLs were maintained from Feb till June to allow for a gradual user transition, and were deprecated on June 1st.
In case you're running SDK version prior to v2.42 then you'll need to retrofit those changes manually.

MartyG-RealSense, 2021-06-28:
@ev-mp   Thanks very much for the answer!

@ankistr  The information provided by @ev-mp is correct.

closed 2021-07-11 by MartyG-RealSense