realsenseai/librealsense #9303: T265 return nan

spaltiel, 2021-06-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version | Win10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.45                       | 
| Language                            |  {C++                          | 
| Segment			|  other                          | 

### Issue Description
Main issue - after several minute the T265 return nan instead of pose data.
After getting nan we tried to open RealSense viewer and we get an error (see the attached screen shoot).
T265 and D435 connected through hub.
Same issue occur both with USB3 and USB2.

We followed rs-tracking-and-depth.cpp example. Up to getting the nan everything is as expected and no issue occur.
How we can reestablish connection and pose data t the T265 device?

[image]

RealSenseCustomerSupport, 2021-07-06:
----------------------------------------------

We are aware of this NaN issue of T265. We highlight it on our errata: https://www.intelrealsense.com/download/16547/?_ga=2.135304565.595682092.1625588372-1479303487.1578600965

spaltiel, 2021-07-13:
We understand. However, can you recommend a workaround to recover from this status?

closed 2026-10-23 by MartyG-RealSense