realsenseai/librealsense #9304: Opposite color while using the depth camera by my program.

barryii, 2021-06-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |                                 D435i                      | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                      | 
| Platform                                 | Laptop  |
| SDK Version                            |  5.0.202                         | 
| Language                            |  Python 3.6.12 / OpenCV 3.3.1 by anaconda, cuz Idk why I can't install tensorflow in my computer. | 
| Segment			|  others. Actually, I have no idea what's should be filled here.                          | 

### Issue Description
First, I'm not sure is this sdk that required info meant.
[image]

And my problem is when I use my program to run D435i's depth camera, its color will be the opposite of Intel RealSense Viewer v2.40.0. The color scheme is Jet. I saw there's an issue that's highly similar with my problem (#3363 ), but the way they provided below I've known and tried and it's unhelpful for my problem. Or is there any other issue that's similar with my issue, please let me know!

Intel RealSense Viewer's picture:
[image]

my program's:
[image]

It's totally opposite. I need the top one's effect.
my program:
```py
import numpy as np
import cv2
import os
import tensorflow as tf
import pyrealsense2 as rs
from datetime import datetime

pipeline = rs.pipeline()
config = rs.config()
# depth_sensor=rs.depth_sensor()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
cfg=pipeline.start(config)
colorizer=rs.colorizer()
colorizer.set_option(rs.option.color_scheme, 0)
dev = cfg.get_device()
depth_sensor = dev.first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 5)
colorizer.set_option(rs.option.visual_preset, 0) # 0=Dynamic, 1=Fixed, 2=Near, 3=Far
colorizer.set_option(rs.option.min_distance, 0)
colorizer.set_option(rs.option.max_distance, 16)
rs.option.histogram_equalization_enabled
while True:
	now = datetime.utcnow() # current date and time
	print(now)
	date_time = now.strftime("%H_%M_%S") #now.strftime("%Y_%m_%d_%H_%M_%S_%f")
	frames = pipeline.wait_for_frames()
	depth_frame = frames.get_depth_frame()
	color_frame = frames.get_color_frame()
	# depth_frame_distance=depth_frame.get_distance()
	depth_sensor_scale=depth_sensor.get_depth_scale
	depth_color_frame=colorizer.colorize(depth_frame)
	if not depth_frame or not color_frame:
		continue
	depth_image = np.asanyarray(depth_color_frame.get_data())
	color_image = np.asanyarray(color_frame.get_data())
	cv2.imwrite(f'D:\\a\\color_image_{date_time}_{loop}.jpg',color_image)
	im = np.vstack([depth_image, color_image])
	cv2.imwrite(f'D:\\a\\im{date_time}_{loop}.jpg',im)
	cv2.imwrite(f'D:\\a\\depth_image_{date_time}_{loop}.jpg',depth_image)
	
	loop=loop+1
	# if loop==30: break
	cv2.namedWindow('RealSense', cv2.WINDOW_NORMAL)
	cv2.imshow('RealSense', im)
	# Stop if escape key is pressed
	k = cv2.waitKey(30) & 0xff
	if k==27:
		break
pipeline.stop()
cv2.destroyAllWindows()
```

Are there any ideas or something else I could provide? Please tell me!

ev-mp, 2021-06-28:
@barryii , I've noticed that in the snapshots the Blue and Red hues are reversed, which leads me to think that the root-cause is in 
```python
        depth_image = np.asanyarray(depth_color_frame.get_data())
	cv2.imwrite(f'D:\\a\\color_image_{date_time}_{loop}.jpg',color_image)
```
The CV's default format for color-space is BGR and not RGB ( as you've correctly configured in `config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)`)

When you extract the data of colorized depth map it is in RGB format, but then you save it using `cv2.imwrite(..` that expects the data to be BGR-ordered...

MartyG-RealSense, 2021-06-28:
Hi @ev-mp  I reached the same conclusion at the exact time as you   :)   Thanks again.   I will post my full answer to @barryii anyway!

Hi @barryii  Your problem does seem to be similar to https://github.com/IntelRealSense/librealsense/issues/3363

The RealSense Viewer applies a range of depth colorization settings by default, whilst applications programmed by RealSense users do not apply these defaults.  Post-processing filters are also not applied by default in a new script.  This means that a depth image generated by a Python script can appear noticably different from a RealSense Viewer image unless colorization and post-processing settings are deliberately programmed into that script to replicate the Viewer's settings.

The link below provides a range of information resources for programming colorization resources in a Python application.

https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671

In the list of color schemes linked to in that list of resources, Jet is '0', which you have correctly defined in your script with **colorizer.set_option(rs.option.color_scheme, 0)**.  You have also set the color to Dynamic like the Viewer with **colorizer.set_option(rs.option.visual_preset, 0)**

The way that you have set up **depth_image = np.asanyarray(depth_color_frame.get_data())** also looks correct to me.

A reversal of the colors like the one in the above images can occur due to the OpenCV **color space** definition, because the OpenCV default for color is **BGR**.

https://github.com/IntelRealSense/librealsense/issues/4542

barryii, 2021-06-29:
Hi, @ev-mp, @MartyG-RealSense. OMG thank you guys so much!!!
I use this to solve my problem at last. I only have to convert the depth image, the color image one is no problem from the beginning so. For other people as a reference.
```py
depth_image = cv2.cvtColor(depth_image, cv2.COLOR_RGB2BGR)
```

MartyG-RealSense, 2021-06-29:
You are very welcome @barryii - thanks for sharing your solution with the RealSense community!

closed 2021-06-29 by barryii