realsenseai/librealsense #9305: Distortion when color and depth information are combined.

DanielSchimpf, 2021-06-28:
Hello,

When I use this version of the realsense viewer the color infomration does not get projected correctly to the depth information. 
The two pictures should explain what I mean. The first picture shows the 3D visualization and one can see that the piece of paper and everything else gets streched into the area where no information seems to be available. 
The second picture shows the two streams seperated. 
[image]
[image]

What causes this distortion? If I run the realsense viewer on windows it does not happen... In windows the depth map gets cropped (because I believe the sensors have different FOV), but just a little. 

Thanks!

| | |
|---|---|
|**librealsense**|2.47.0 RELEASE|
|**OS**|Linux| Ubuntu 16
|**Name**|Intel RealSense D435|
|**Serial Number**|021122070656|
|**Firmware Version**|05.12.14.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|939423025388|
|**Firmware Update Id**|939423025388|

Please provide a description of the problem

MartyG-RealSense, 2021-06-28:
Hi @DanielSchimpf  Could you use the instructions in the link below to check whether you have a pair of **GLSL** options called 'Rendering' and 'Processing' under the Performance section of the **Settings** interface of the RealSense Viewer, and disable these two options if they are present, please.   

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

If you are able to disable these two GLSL options, does it improve the performance of 3D mode in your Viewer in Ubuntu, please?

DanielSchimpf, 2021-06-28:
Hey, both options were already disabled, so no change.

ev-mp, 2021-06-28:
@DanielSchimpf , the RGB FOV for D435 is smaller than for the Depth , therefore the OpenGl renderer wraps RGB texture over the regions with no coverage. 
The artifact has no impact on the actual point cloud data and can be safely disregarded. Alternatively, you can modify the Realsense-Viewer code and prevent the usage of OpenGl texture wrap filters.

Duplicate of #3106, #2237

closed 2021-07-11 by MartyG-RealSense