realsenseai/librealsense #9307: How to get aligned L515 dataset？

JZhu1208, 2021-06-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   |v2.45.0| 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC |
| SDK Version                            | v2.45.0                       | 
| Language                            |  C++ matlab                         | 
| Segment			|                    | 

Hello, I saw a well-aligned color image and depth image at https://www.intelrealsense.com/lidar-camera-l515/, how did you get it?

Or does Intel RealSense L515 provide a dataset?  Thank you very much.
 
[image]

MartyG-RealSense, 2021-06-29:
Hi @JZhu1208   You can align depth and color with your L515 using a pre-made example program called **rs-align**.  If you have installed the RealSense SDK on Windows then you can find a ready-built executable version of this example by right-clicking on the RealSense Viewer launch icon and selecting the **Open file location** menu option.  This opens the **Tools** SDK folder where launchable .exe example files are located.

Find the rs-align program in the list of files in the folder, ensure that your L515 camera is plugged in and then launch the program with a double-leftclick.

[image]

[image]

MartyG-RealSense, 2021-06-29:
You should be able to save an aligned image using the OpenCV **imwrite** instruction, like in the L515 case about creating aligned-image files in the link below.

https://github.com/IntelRealSense/librealsense/issues/7081

MartyG-RealSense, 2021-07-06:
Hi @JZhu1208  Do you require further assistance with this case, please?  Thanks!

JZhu1208, 2021-07-06:
Thank you for your help. Regarding this question, I don’t think I need help anymore. You could close this issue. Thank you!

MartyG-RealSense, 2021-07-06:
Thanks very much @JZhu1208 - you are very welcome  :)    As suggested, I will close the case.  Thanks again!

closed 2021-07-06 by MartyG-RealSense