realsenseai/librealsense #9308: L515 stops working after 24/48 hours

AJMena, 2021-06-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.04.01) | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.40.0.2482 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
We have running three L515 during a long period of time. All of them are connected to the same PC using USB 3.2. After 24/48 hours one of them stops working and gives an exception calling “rs2.pipeline.wait_for_frames”. We checked the L515 in the realsense viewer. The depth sensor shows us a white screen and no frames are receiving error. The RGB camera keeps working fine. This happened two times in a row after about 24/48 hours and both times the same sensor failed.
We are now checking the others two L515 alone. 

We do not know why it stops working.  Any idea?

MartyG-RealSense, 2021-06-29:
Hi @AJMena  Does the affected L515 work normally after the camera is reset until the next time that it happens after 24/48 hours, please?

If you are able to use this camera, could you check its ASIC and Projector temperatures in the **Controls** category of the Stereo Module options in the RealSense Viewer's side-panel and let me know what its temperature values are once the camera has been running for about 10 minutes.

AJMena, 2021-06-29:
Hi MartyG-RealSense.

After reseting the camera it works fine.

Temperatures after turning on:
[image]

Temperatures after 10 minutes running:
[image]

MartyG-RealSense, 2021-06-29:
The temperatures are noticably increasing in all categories by an average of 7 degrees C after 10 minutes.  Page 17 of the current edition of the data sheet document for the L515 model recommends an official maximum of 30 degrees during operation.  

https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet

[image]

Is the camera being used in a location where there is a high environmental air temperature, please?

AJMena, 2021-06-30:
The sensors are used in an interior, the environmental is normal.
We checked an all three sensor and all of them are getting their temperatures up gradually.
Is there any way to prevent the temperature from going up?

MartyG-RealSense, 2021-06-30:
If all the L515 are being used on the same computer and their temperature is gradually increasing constantly over time instead of reaching a certain point and then levelling out, I wonder if there could be a glitch on the computer's USB ports that is causing the cameras to heat up.  

Could you try starting the cameras from a cold state and then touching their casing after a few minutes to see whether they have become hot to the touch?

AJMena, 2021-07-02:
We checked the viewer after a failure. We get a white screen with the error "No frames received!" in the depth and infrared frame. But the color frame keeps working fine.
We restarted the computer and checked the viewer again. The depth and infrared frames keep failing. The color frame works fine.
We use the depth and infrared in our program, but not the color.
We think that after a "failure" the sensor keeps working fine. The problem is that the streams were reserved or something like that. Then that reserved connection was lost so now nobody can access it.
Restarting the L515 solves the problem but we want to prevent the failure. We want to know if it could be possible to “access again” the streams after this type of failures.
We updated the SDK and the firmware to the latest version, but nothing changed.

AJMena, 2021-07-02:
We also tested the L515 in the office using the viewer. After 2 hours of continuous working the temperatures are between 55-65. That cannot be good. What is the expected temperature of the L515 after some hours of working?
[image]

MartyG-RealSense, 2021-07-02:
The data sheet documentation for the L515 model only specifies a maximum recommended **ambient** temperature, not values for the individual temperature values.  Ambient temperature typically refers to the temperature of the air.

I ran a test with my own L515 and observed that the LDD (laser driver) temperature progressively increased over a 10 minute period even in a cool-aired room and approximately matched the rate of increase over the same time period that your own camera demonstrated in https://github.com/IntelRealSense/librealsense/issues/9308#issuecomment-870527478  

So I do not believe that the same rules for judging appropriate temperature that are applied to the 400 Series stereo depth cameras can be applied to the L515 lidar camera, and that the air temperature and casing temperatures specified in the data sheet are the main temperature benchmarks that the camera should be judged by.

As the camera is able to stream normally once reset, this indicates that there is not a fault in the camera hardware and that it is more likely to be a software issue or a glitch on the computer's USB port.

If you receive an exception error then you should be able to program a mechanism to catch that error and automatically take an action in response, such as performing a **hardware_reset()** instruction to reset the camera without human intervention.  The SDK's**Error Handling Scheme** documentation provides information about setting up an exception catcher in C++.

https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme

AJMena, 2021-07-02:
So, there is no guidelines for the temperatures of the device, is that correct?

I'm trying to implement the hardware_reset(). I am unable to know how much I must wait for the device to be ready. I did not find any examples. Any suggestions?

MartyG-RealSense, 2021-07-02:
You can view what each temperature represents in the RealSense Viewer by hovering the mouse cursor over the question-mark icon at the end of each temperature reading.  I do not have guidelines available for those temperatures on L515 though.

It can take 2 to 3 seconds to reset a RealSense camera.  A 5 second sleep time should therefore be a reasonable value.

The link below has a C++ multicam hardware_reset() script that iterates through all attached cameras.  This may be a useful reference for you.

https://community.intel.com/t5/Items-with-no-label/D435-streams-work-once-then-crash/m-p/499732#M6053

MartyG-RealSense, 2021-07-08:
Hi @AJMena   Do you require further assistance with this case, please?  Thanks

AJMena, 2021-07-09:
You can close it.
We didn't find any useful information here to solve the problem.
Thank you anyway.

closed 2021-07-09 by AJMena