realsenseai/librealsense #9310: Question about the measure example

Rad0n1, 2021-06-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            |   2.45.0   | 
| Language                            |  c++ | 
| Segment			|  Computer vision  | 

Hey, this might be a stupid question.

I am using a background subtraction algorithm to find a new object, and then want to use the depth camera measurement example to create a way to find the measurements of the object placed in front of the camera. The problem I am running in to with the example that it has a lot of black spots of where the depth data is missing because the depth and colour are aligned. Now was my question if there is a possibility to get a seperate colour image to run through the algorithm.

I do realise there is a point to be made for not needing this, as if there is no depth data, it won't be possible to get measurements in the first place.

I am mostly just trying to learn what is possible and I am not sure where to look for this specific problem as I don't know what combines the 2 images.

MartyG-RealSense, 2021-06-29:
Hi @Rad0n1   An easy way to fill in black gaps on an image where color is mapped to depth would be to apply a post-processing **Hole-Filling Filter**.  The image below from the RealSense Viewer's 3D point cloud mode demonstrates a scene before (upper) and after (lower) when applying the hole-filling filter.

[image]

The Intel documentation link below provides information about how the Hole-Filling filter works.

https://dev.intelrealsense.com/docs/post-processing-filters#section-holes-filling-filter

In regard to the mechanism for combining images in the **rs-measure** example, that program uses **color to depth** alignment rather than the more common depth-to-color alignment.  My understanding is that it was programmed this way to provide an example of programming color-to-depth alignment and that it is not vital to align images this way for the purposes of point-to-point measurement.  For example, a Python conversion of rs-measure created by a RealSense community member used depth-to-color instead.

In color-to-depth alignment, the field of view (FOV) size of the color is resized to the FOV size of the depth.  On camera models where the FOV size of depth and color are different (like the D435 and D435i), this can stretch the aligned image out to fill the full size of the panel because the depth sensor on these models has a larger FOV size than the RGB sensor.  

When performing **depth-to-color**, the depth FOV is resized to the color FOV size, so on D435 / D435i the outer regions of the aligned image can become excluded from the image due to the color sensor having a smaller FOV.

If you would prefer to perform depth-color alignment in OpenCV, the **rs-grabcuts** example program may suit your requirements.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts

Rad0n1, 2021-06-29:
That's exactly what I was looking for! Thank you so much for your detailed answer, it's really appreciated 😄

closed 2021-06-29 by Rad0n1

MartyG-RealSense, 2021-06-29:
That's great to hear, @Rad0n1 - I'm pleased that I could help!