realsenseai/librealsense #9313: SR300 Stuck in Recovery mode

Mohali2609, 2021-06-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | SR300 | 
| Firmware Version                    | 3.26.10 | 
| Operating System & Version   |   {Win 10 |
| Platform                                 | PC |
| SDK Version                            |  2.47.0 | 
| Language                            | C | 
| Segment			| others | 

I have an in build SR300 intel realsense camera in my laptop. ever since Windows 10 update, windows hello stopped working. It was no issue to me since I don't use Windows hello. But today Windows update did something again and now the camera is stuck in Recovery mode. I can't exit using the rs-fw-update tool (or I am using it wrong) and I can't update the firmware to any known firmware 3.10.10, 3.17.0 or even 3.26.10 due the device being locked. I read somewhere that after 10 or so FW updates the device locks down. 

Note: I am willing to install Linux distro (Ubuntu or Kubuntu) to test out anything.

What can I do?

Thanks

MartyG-RealSense, 2021-06-29:
Hi @Mohali2609  The **rs-fw-update** tool is for RealSense 400 Series cameras and not for the SR300 model.

Up to 20 **downgrades** of the firmware to an earlier version than currently installed are permitted.  Upgrades to newer versions do not add to the counter and when the limit of 20 is reached firmware updates are still permitted, though only to a newer version than currently installed.

If you believe that the camera's functionality was broken by Windows updates then you can 'roll back' updates to attempt to return the computer's configuration to an earlier state.

https://www.howtogeek.com/235474/how-to-roll-back-builds-and-uninstall-updates-on-windows-10/

If the camera was indeed rendered non-functioning by Windows updates then even if a rollback fixes the problem then it may occur again if automatic updates continue to be permitted.  You can therefore delay the installation of updates for a time.  More information about this can be found in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000300941/comments/360014057354

A last-resort option would be to perform a recovery installation of Windows if your particular computer supports it.  Some laptop brands have an image of the computer's original as-new configuration stored on them, including all the programs and files that the computer came with.  Using such a restoration image is different to a normal wipe and reinstall of Windows, as it restores the computer to how it was when purchased.   If the computer has had its hard drive wiped during its lifetime though then the recovery image may no longer be available.

Another option would be to wait until a future Windows update to see whether something changes in Windows that restores the camera as a side-effect.  This has happened before in the past with the SR300.

I would emphasize that a built-in SR300 should not be treated the same as an external USB-attached SR300.  Built-in cameras are much more difficult to correct once they develop a problem.  It is also recommendable that the technical support site of the computer's manufacturer is used rather than the Intel RealSense support forums, since solutions such drivers that work for a USB SR300 may not be suited to the special configuration of a built-in SR300.

Mohali2609, 2021-06-29:
Hi @MartyG-RealSense 

Thanks for tre fast respons. 

You correct in-built version should be treated differently than external ones but sadly windows didn’t think so. It went ahead and updated it any way... and I have tried to roll back the update the update, didn’t help. I even tried to install the manufacturer driver (from Acer by the way and very very old) but it nevers works. It just throws a an error. And through real view update firmware function I get the that I need to install a newer firmware than 3.26.1.

Mohali2609, 2021-06-29:
@MartyG-RealSense I have been look around for any firmware that is newer than 3.26.1 and while i didn't find the file, I found 

this https://github.com/IntelRealSense/librealsense/issues/4833

and in the build code error it mentions SR3XX_FW_Image-3.27.3.0.bin.

would it be possible to compile the build and pull the firmware from there?
I don't mind if the camera becomes SR305 as long as I get some of the functionality back. (I don't use Windows Hello anyway)

MartyG-RealSense, 2021-06-29:
I have an old original USB SR300 that is now recognized as an SR305 in the RealSense Viewer after a firmware update was performed on it in the Viewer a while ago.  I have heard that making an SR300 into an SR305 may break compatibility with very old SR300 applications from around 2016.  My SR300 works fine in the RealSense SDK in SR305 mode though.

I do not know if a built-in SR300 would be able to be recognized by the RealSense Viewer.  I would speculate that if it was not listed automatically in the side-panel of the Viewer after it is launched then it might be findable using the **Add Source** button at the top of the side-panel (which can detect the built-in RGB webcams of laptops).  I have not personally tested it myself with a computer with a built-in SR300 though.

[image]

Mohali2609, 2021-06-29:
RealSense was and is able to recognize the camera automatically. I have managed to pull the 3.27.30 file and upload it to the camera now it is recognized as 305 instead of 300 but everything is working... thank you very much for the help.

Do you want me to link the firmwares.bin files

they from intel, finding them though was pain.

MartyG-RealSense, 2021-06-29:
That's excellent news that you were able to restore your SR300 to working order!

As long as the source of the firmwares that you found is Intel then it should be fine to share the link.  I always strongly discourage people from going searching round dark corners of the internet for old drivers because of the risk of picking up malware and viruses.

Mohali2609, 2021-06-29:
Okay..
They are from intel I believe, if not tell me and I will edit the post..

3.26.1.0:
 https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin

3.27.3
https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.27.3.0.bin

3.27.3 wasn't listed or mentioned anywhere else except the in the issue I mentioned above and when I tried to modify the link
it worked and download a bin file that matches the file size of 3.26.1.0, and as you can see it works.

MartyG-RealSense, 2021-06-29:
**https://librealsense.intel.com** is a valid Intel file respository address, yes.  Thanks very much.

[image]

MartyG-RealSense, 2021-07-05:
Hi @Mohali2609  Do you require further assistance with this case, please?  Thanks!

Mohali2609, 2021-07-05:
@MartyG-RealSense 

No, the camera is working fine after the firmware update. I will close the case now..

thanks

MartyG-RealSense, 2021-07-05:
Excellent news @Mohali2609 - thanks very much for the update   :)