realsenseai/librealsense #9318: Delay of data depth

anhTuan0712, 2021-06-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   | 1.05.05.00| 
| Operating System & Version |  Win (8.1/10)  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |                         | 
| Language                            |  {C#                         | 
| Segment			|                      | 

### Issue Description
## My source code
## Config section
[image]
## Get data and save to file
[image]
[image]
## Main 
[image]
## The full source code can be viewed here: https://drive.google.com/drive/folders/1IgWd7_hAViMnLIaFwaeZFWV6dKGiRuSQ?usp=sharing
### Problem phenomenon
## when using "Align to Depth" mode => depth data is not DELAY
[image]
##  when not using "Align to Depth" mode => depth data is DELAY
[image]
## The full data can be viewed here: https://drive.google.com/drive/folders/1IgWd7_hAViMnLIaFwaeZFWV6dKGiRuSQ?usp=sharing
### May be similar to this probleme https://github.com/IntelRealSense/librealsense/issues/8710
Can anyone give me a solution to this problem.
Hi @RealSense-Customer-Engineering , Hi @RealSenseSupport .
I would like to thank you for your interest in this matter of mine.

MartyG-RealSense, 2021-06-30:
Hi @anhTuan0712   Comparing the two sets of images, my understanding of the problem is that the depth frame is delayed compared to the infrared and color frames when alignment is not being used.  This is shown by the arm position in the depth frames being time-delayed compared to the correct arm position in the infrared and color frames.  Is this correct, please?

I note that in the **Config** section of your code you are setting the color resolution to 640x480.  This resolution is not supported on the L515 camera model.  The supported L515 RGB color resolutions are 960x540, 1280x720 or 1920x1080, as described in the extract below from the L515 data sheet document below.

[image]

How does it perform in non-aligned form if color is set in the cfg code to 960x540 (the nearest resolution to 640x480).

anhTuan0712, 2021-07-01:
> Hi @anhTuan0712 Comparing the two sets of images, my understanding of the problem is that the depth frame is delayed compared to the infrared and color frames when alignment is not being used. This [...]
Hi  @MartyG-RealSense.
Yes that's right, please explain this to me.

anhTuan0712, 2021-07-01:
> I note that in the **Config** section of your code you are setting the color resolution to 640x480. This resolution is not supported on the L515 camera model. The supported L515 RGB color [...]

Well thanks for your advice @MartyG-RealSense .

MartyG-RealSense, 2021-07-07:
Hi @anhTuan0712  Do you require further assistance with this case, please?  Thanks!

anhTuan0712, 2021-07-08:
Hi @MartyG-RealSense. I am looking forward to hearing from you.

MartyG-RealSense, 2021-07-08:
In my earlier advice I stated that 640x480 was not a supported resolution on L515 for the **color** stream.

What further advice do you require please?

anhTuan0712, 2021-07-14:
> Hi @anhTuan0712 Comparing the two sets of images, my understanding of the problem is that the depth frame is delayed compared to the infrared and color frames when alignment is not being used. This [...]

this is my main problem, please explain it.

MartyG-RealSense, 2021-07-14:
I looked again at your code and focused on the section of 'else' code that apparently tells the program what to do if align = false (since you are having problems with depth delay when alignment is not being used.

[image]

I note from the above code that you are disposing of infrared and color if NiR is not false but there is not a disposal line for depth.  Are you disposing of depth frames too?  Non-disposal of frames could cause .NET to become jammed up with frames, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923

anhTuan0712, 2021-07-20:
> Are you disposing of depth frames too? Non-disposal of frames could cause .NET to become jammed up with frames, as described in the link below.

Hi @MartyG-RealSense, 
My problem was solved following this suggestion of yours. thank you very much!

MartyG-RealSense, 2021-07-20:
Great to hear that the advice resolved your problem!  I will close this case as you have achieved a solution.  Thanks again!

closed 2021-07-20 by MartyG-RealSense