realsenseai/librealsense #9319: Error running realsense-viewer

jalapatip, 2021-06-30:
@MartyG-RealSense, @c-mendoza,  @dyamized, and @RealSenseSupport

I'm trying to install librealsense on M1 based mac and ran into some issues. I've been following the threads [issue](https://github.com/alicevision/geogram/issues/2#issuecomment-536835320) and [#8090](https://github.com/IntelRealSense/librealsense/issues/8090) and could successfully build the project so far. But am not able to run realsense-viewer. I additionally got the below error which I fixed by adding 'set(CMAKE_MACOSX_RPATH 1)' to the CMakeLists.txt file.

MACOSX_RPATH is not specified for the following targets: realsense2-gl

Current error:
'zsh: command not found: realsense_viewer'

I tried building the project using the XCode but it did not work.

Any help would be appreciated. Thanks in advance!!

MartyG-RealSense, 2021-06-30:
Hi @jalapatip  there was a past case in which a RealSense MacOS user experienced the **MACOSX_RPATH is not specified for the following targets: realsense2-gl** error.  The solution that was suggested, which worked for them, was to remove **-G Xcode** from the CMake build instruction if you are using that build command.

https://github.com/IntelRealSense/librealsense/issues/5258#issuecomment-554918028

jalapatip, 2021-06-30:
> Hi @jalapatip There was a past case in which a RealSense MacOS user experienced the **MACOSX_RPATH is not specified for the following targets: realsense2-gl** error. The solution that was suggested, which worked for them, was to remove **-G Xcode** from the CMake build instruction if you are using that build command.
> 
> [#5258 (comment)](https://github.com/IntelRealSense/librealsense/issues/5258#issuecomment-554918028)
> 
> Using this method may be the appropriate way to deal with that error.

Hello @MartyG-RealSense, thanks for the update. I did try without '-G Xcode' and the project compiled successfully. But I still couldn't open realsense-viewer.

MartyG-RealSense, 2021-06-30:
Have you included **-DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true** in the CMake build instruction, please?  

These commands build the SDK's example programs and tools, including the RealSense Viewer.

jalapatip, 2021-06-30:
The following is the command that I'm using: 
cmake .. -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false

I also tried clearing the cache and re-building.

MartyG-RealSense, 2021-06-30:
Could you check the location where the examples and tools are stored to confim that realsense-viewer is present there please.  You should be able to find it by doing a file search for **realsense-viewer** in MacOS.

https://support.apple.com/en-ie/guide/mac-help/mchlp2305/mac

jalapatip, 2021-06-30:
I see examples and tools in the following locations:

1. /librealsense/tools/realsense-viewer
2. /librealsense/examples
3. /librealsense/build/tools/realsense-viewer
4. /librealsense/build/examples

MartyG-RealSense, 2021-06-30:
Please go to the **librealsense/build/tools/realsense-viewer** folder and see if you can launch realsense-viewer from that folder instead of using the 'realsense-viewer' command.

jalapatip, 2021-06-30:
I see a 'RealsenseToolsRealSenseViewer.xcodeproj' file in the folder and I tried to build and run it in XCode but it did not help. Anything else that I should be doing?

MartyG-RealSense, 2021-06-30:
I have seen the **zsh: command not found: realsense_viewer** a couple of times in cases involving the Arm-based Nvidia Jetson and Ubuntu but not on MacOS.  

I note from your CMake instruction that you are not telling it to build in **Release** mode, so it may be building by default in **Debug** mode instead.  Could you try including the command below in your CMake instruction to build librealsense in Release mode please?

**-DCMAKE_BUILD_TYPE=Release**

jalapatip, 2021-06-30:
Thanks, I just tried adding '-DCMAKE_BUILD_TYPE=Release'. Unfortunately, it is still not working. 

But when I repeated the above steps where I set the target as realsense-viewer and executed 'RealsenseToolsRealSenseViewer.xcodeproj', I'm only getting the following error:

clang: error: no such file or directory: '/Users/jalapatip/Work/DRCL/librealsense/build/libcurl/libcurl_install/lib/libcurl-d.a'
Command Ld failed with a nonzero exit code

I see that the 'librealsense2-gl.2.47.0.dylib' file in the librealsense/build/src/gl/Debug folder.

MartyG-RealSense, 2021-06-30:
Okay, let's forget about **RealsenseToolsRealSenseViewer.xcodeproj'** as it is likely not related to the problem in this case.  I do not have a Mac computer to test with at my location, so I very much appreciate your patience.

 Are you able to launch any built SDK programs other than the Viewer?

jalapatip, 2021-06-30:
No problem, thank you so much for the help!!

I could run the examples and they are working fine.

MartyG-RealSense, 2021-06-30:
Thanks very much!   If you are able to run the **rs-capture** example then that would likely rule out the possibility of the problem being related to OpenGL graphics, as the Viewer and rs-capture both make use of OpenGL. 

Have you tested the **Depth Quality Tool** successfully, as this has a similar design to the Viewer tool.

jalapatip, 2021-06-30:
Yes, I could run the rs-capture example and it is working fine.

MartyG-RealSense, 2021-06-30:
I note that you already tried cleaning the CMake cache as https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-837115091 suggested regarding an error with **libcurl_install/lib/libcurl-d.a** but this did not correct the problem for you.  In the comment below that link, the RealSense user thought that they had already performed the clean but repeating the clean removed the problem.

jalapatip, 2021-06-30:
I tried cleaning the cache and when it did not work, I deleted the build folder and re-built the package just to be sure. It did not fix the issue for me. Also, I see that the build is for arm64. 

Link realsense-viewer (arm64) 0.1 seconds

Ld /Users/jalapatip/Work/DRCL/librealsense/build/tools/realsense-viewer/Debug/realsense-viewer normal (in target 'realsense-viewer' from project 'RealsenseToolsRealSenseViewer')
    cd /Users/jalapatip/Work/DRCL/librealsense/tools/realsense-viewer
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -target arm64-apple-macos11.3 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -L/Users/jalapatip/Work/DRCL/librealsense/build/tools/realsense-viewer/Debug -F/Users/jalapatip/Work/DRCL/librealsense/build/tools/realsense-viewer/Debug -filelist /Users/jalapatip/Work/DRCL/librealsense/build/tools/realsense-viewer/RealsenseToolsRealSenseViewer.build/Debug/realsense-viewer.build/Objects-normal/arm64/realsense-viewer.LinkFileList -Xlinker -rpath -Xlinker /Users/jalapatip/Work/DRCL/librealsense/build/src/gl/Debug -Xlinker -rpath -Xlinker /Users/jalapatip/Work/DRCL/librealsense/build/Debug -Xlinker -rpath -Xlinker /opt/homebrew/lib -Xlinker -object_path_lto -Xlinker /Users/jalapatip/Work/DRCL/librealsense/build/tools/realsense-viewer/RealsenseToolsRealSenseViewer.build/Debug/realsense-viewer.build/Objects-normal/arm64/realsense-viewer_lto.o -Xlinker -no_deduplicate -L/usr/local/opt/openssl@1.1/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names /Users/jalapatip/Work/DRCL/librealsense/build/common/fw/Debug/libfw.a -framework OpenGL -framework OpenGL /Users/jalapatip/Work/DRCL/librealsense/build/src/gl/Debug/librealsense2-gl.2.47.0.dylib /Users/jalapatip/Work/DRCL/librealsense/build/Debug/librealsense2.2.47.0.dylib /opt/homebrew/lib/libglfw.3.3.dylib /Users/jalapatip/Work/DRCL/librealsense/build/libcurl/libcurl_install/lib/libcurl-d.a /opt/homebrew/opt/openssl@1.1/lib/libssl.dylib /opt/homebrew/opt/openssl@1.1/lib/libcrypto.dylib -Xlinker -no_adhoc_codesign -Xlinker -dependency_info -Xlinker /Users/jalapatip/Work/DRCL/librealsense/build/tools/realsense-viewer/RealsenseToolsRealSenseViewer.build/Debug/realsense-viewer.build/Objects-normal/arm64/realsense-viewer_dependency_info.dat -o /Users/jalapatip/Work/DRCL/librealsense/build/tools/realsense-viewer/Debug/realsense-viewer

clang: error: no such file or directory: '/Users/jalapatip/Work/DRCL/librealsense/build/libcurl/libcurl_install/lib/libcurl-d.a'
Command Ld failed with a nonzero exit code

MartyG-RealSense, 2021-06-30:
And you have followed the suggestion in https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-836892782 to modify CMake/external_libcurl.cmake to add the line below after line 35 of that file?

**-DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl**

jalapatip, 2021-06-30:
Yes

set(CURL_FLAGS -DBUILD_CURL_EXE=OFF -DBUILD_SHARED_LIBS=OFF -DUSE_WIN32_LDAP=OFF -DHTTP_ONLY=ON -DCURL_ZLIB=OFF -DCURL_DISABLE_CRYPTO_AUTH=ON -DCMAKE_USE_LIBSSH2=OFF -DBUILD_TESTING=OFF -DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl)

MartyG-RealSense, 2021-06-30:
I believe that at the tine of the https://github.com/IntelRealSense/librealsense/issues/8090 case, the SDK version being used was **2.44.0**.  So 2.44.0 could be tested with the Mac M1 procedure to see whether the problem is one introduced since 2.44.0.

jalapatip, 2021-06-30:
Hey, reverting to 2.44 worked!! Thank you so much for the help!!

closed 2021-06-30 by jalapatip

MartyG-RealSense, 2021-07-01:
That's excellent news, @jalapatip - thanks very much for the update!

chiawenwang, 2024-01-28:
> Hey, reverting to 2.44 worked!! Thank you so much for the help!!

Hi! I also meet the problem like yours "zsh: command not found: realsense_viewer". I wonder how can I check and revert the version of SDK?