realsenseai/librealsense #9322: ponitcloud（3D viewer）no show

JC-FOSU, 2021-07-01:
| | |
|---|---|
|**librealsense**|2.47.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0461386|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003ba67ef62|
|**Firmware Update Id**|0003ba67ef62|

Please provide a description of the problem

MartyG-RealSense, 2021-07-01:
Hi @JC-FOSU   Could you check in the Viewer's settings whether a pair of **GLSL** options are enabled, and disable them both if they are enabled.  Instructions for doing so are in the link below.  Is the point cloud generated if these options are disabled, please?

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

JC-FOSU, 2021-07-01:
good job! thank u very much!

#8110 (comment)

MartyG-RealSense, 2021-07-01:
You are very welcome  :)

MartyG-RealSense, 2021-07-08:
Hi @JC-FOSU   Do you require further assistance with this case, please?  Thanks

closed 2021-07-14 by MartyG-RealSense