realsenseai/librealsense #9326: About the integration of using two D435+T265+rplidar

ju-mingyue, 2021-07-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |  Linux (Ubuntu18.04.5)  | 
| Kernel Version (Linux Only)    |  4.9.201                                     | 
| Platform                                 | NVIDIA JetsonNano B01 |
| SDK Version                            |2.41.0                        | 
| Language                            | ROS packages                           | 
| Segment			|  Robot                         | 

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |  Linux (Ubuntu18.04.5)  | 
| Kernel Version (Linux Only)    |  4.9.201                                     | 
| Platform                                 | NVIDIA JetsonNano B01 |
| SDK Version                            |2.41.0                        | 
| Language                            | ROS packages                           | 
| Segment			|  Robot                         | 

### Issue Description
Hello, I encountered the following problems when using two D435 and one T265 and one rpliadr for project development.
1. After I adjusted the combination of the two D435 and T265 programs, I hope to use the rtabmap program to combine to convert the depth information into raster map information, but my two D435 cameras published different topics, and I put them separately The two D435 cameras are named D435_L and D435_R. Their topics are D435_L/depth/image_rect_raw, D435_L/color/image_raw, D435_L/color/camera_info, D435_R/depth/image_rect_raw, D435_R/color/image_raw, D435_R/color/ camera_info. And rtabmap only accepts three topics D435/depth/image_rect_raw, D435/color/image_raw, D435/color/camera_info, so I want to merge the topics published by D435_L and D435_R. I don't know how to modify the rtabmap program so that it can accept the topic of two D435 cameras.
2. I don’t know how to add the obstacle information converted from the depth information to the local map of the robot’s autonomous navigation.

ju-mingyue, 2021-07-02:
@@Combinacijus @MartyG-RealSense

Combinacijus, 2021-07-02:
Not related to the problem but I have suggestion to include error message in text format too it's easier to copy and search that way. Images are ok for nice colored formatting but that's that. Also for some strange reason images in your question is not rendered could it be because non-english characters in name or maybe because wrong code formatting because triple quotes ` ''' ` are hanging in strange places. For code formatting it should be triple [backticks/backquotes/grave accent](https://en.wikipedia.org/wiki/Grave_accent) ` ``` `

And for the problem I'm not familiar with rtabmap and only starting on using both D435 and T265 cameras together so I can't comment much on that.
Some suggestion would be:
- It's probably possible to combine 2 topic in [rtabmap e.g.](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch) (not sure if it's not old)
- If image messages published from Realsense have Header field (it contains tf frame information) then it should be possible just to [remap](http://wiki.ros.org/roslaunch/XML/remap) both cameras image topics into a single one (single depth, single color topic etc)

> But after running, the error prompted is as follows:
- This might be problem with wrongly configured topic names so double check if Realsense published topics are the same with rtabmap subscribed topics to help with that use `rosrun rqt_graph rqt_graph` to see if topics are connected to nodes

closed 2021-07-21 by ju-mingyue