realsenseai/librealsense #9327: How to set Low Power mood for D435

S-Ashja, 2021-07-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | ( 05.12.12.100 ) | 
| Operating System & Version |   {Linux (Ubuntu 18)| 
| Kernel Version (Linux Only)    |  ( 4.9.140-tegra)                                         | 
| Platform                                 | NVIDIA Jetson Xavier |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {C/python }                          | 
| Segment			|  {others }                          | 

### Issue Description
I am using 4 D345 cameras in config to create a plane using them on Nvidia Xavier.I want to put the devices in low power mode to save power while a session is not active. how I can do that?

MartyG-RealSense, 2021-07-02:
Hi @S-Ashja  If you are using multiple cameras with Python and are controlling them with a single application to create a plane then you may have already developed code for a multi-camera interface.  If you have not then a multicam Python script contributed by a RealSense user in the link below may be a useful reference.

https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras

If you are able to communicate with individual cameras then you can change the cameras' **Laser Power** setting.  By default this setting has a value of '150', whilst its maximum value is '360'.   Setting it to '0' will turn the camera's projector off, whilst setting it to a value greater than 0 should turn the projector back on.

An example of setting the Laser Power in Python (though for a single camera) with the instruction **rs.option.laser_power** can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/8592#issuecomment-800257572

[image]

S-Ashja, 2021-07-02:
@MartyG-RealSense Thank you for your reply.
I will try this and get back here but I wanna know if this would be enough or I would have to turn the laser and the resolution down for power saving

MartyG-RealSense, 2021-07-02:
Typically, just reducing Laser Power is the measure that RealSense users apply to save power.  

C++ has a similar instruction called **RS2_OPTION_LASER_POWER** for setting the Laser Power value.  The SDK has an example librealsense2 C++ script for doing so at the link below.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

You could also stop the camera's pipeline to cease the streaming of data whilst capture is not taking place and then re-start the pipeline just before taking a capture and re-close the pipeline when the capture is completed.  

A RealSense user contributed a C++ script for simultaneously capturing images from multiple cameras.  The script checks for whether the camera model is a D415, so you should edit that section to be D435 instead if you test the script with your D435 cameras.

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468

S-Ashja, 2021-07-02:
@MartyG-RealSense  
Thank you for your help  I think I have sufficient information and can close this issue.

closed 2021-07-02 by S-Ashja

MartyG-RealSense, 2021-07-02:
You are very welcome @S-Ashja  - good luck!