realsenseai/librealsense #9328: How to Export PointCloud.ply of YOLO predefined region WITHOUT using Open3D

yiyangd, 2021-07-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 5.12.13.50 | 
| Operating System & Version |    Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    | 5.4.0-77-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.44.0                        | 
| Language                            |  opencv/python                          | 
| Segment			|  others                           | 

### Issue Description
#2769 https://github.com/IntelRealSense/librealsense/issues/2769#issuecomment-442048035 provide method for using Open3D to export the ply file. However, is it possible for me to use the API provided by pyrealsense2 instead of Open3d ?

I checked the sample code [export_ply_example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py) and noticed there is a function, [rs.save_to_ply](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.save_to_ply.html). I planed to pass the parameter of cropped_frame (captured by YOLO) and its points(i.e. vertices_interest = image_Points[top_cord:bottom_cord,left_cord:right_cord,:].reshape(-1,3)
 in https://github.com/IntelRealSense/librealsense/issues/2769#issuecomment-442047525) into save_to_ply API but the document did not point out what the required data type and shape of parameters for rs.save_to_ply( ) and rs.save_to_ply( ).process( )

Could you please give any suggestions?

Thanks!!

MartyG-RealSense, 2021-07-02:
Hi @yiyangd  I researched your request carefully but could not find a single solution that precisely matched your requirements.  I did find a page of pyrealsense2 scripts that contained separate scripts for image cropping and for exporting a ply using **export_to_ply** instead of save_to_ply.  You may be able to assemble a single script solution using these pieces.

https://www.programmersought.com/article/36975141951/

I recommend performing a browser page 'find' operation for the terms **crop** and **.ply** to find the code sections that are relevant to your needs.

yiyangd, 2021-07-04:
Thanks！

> Hi @yiyangd I researched your request carefully but could not find a single solution that precisely matched your requirements. I did find a page of pyrealsense2 scripts that contained separate scripts for image cropping (with **cv2** instead of YOLO) and for generating a pointcloud that is exported to ply using **export_to_ply** instead of save_to_ply. You may be able to assemble a single script solution using these pieces.
> 
> https://www.programmersought.com/article/36975141951/
> 
> I recommend performing a browser page 'find' operation for the terms **crop** and **.ply** to find the code sections that are relevant to your needs.

MartyG-RealSense, 2021-07-09:
Hi @yiyangd  Do you require further assistance with this case, please?  Thanks!

yiyangd, 2021-07-11:
> Hi @yiyangd Do you require further assistance with this case, please? Thanks!

My question was solved, thanks, again!

MartyG-RealSense, 2021-07-11:
Excellent to hear, @yiyangd - thanks very much for the update!   As you have achieved a solution, I will close this case.

closed 2021-07-11 by MartyG-RealSense