realsenseai/librealsense #9329: Can't receive Depth data - after new sensor initially worked with new firmware.

jkneer, 2021-07-02:
| | |[image]|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Mac OS|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1150534|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003ba0ad033|
|**Firmware Update Id**|0003ba0ad033|

Received  the L515 yesterday. The viewer suggested a firmware update, which I did apply. In the end the confirmation dialog said the camera was updated. But then the camera "did not reconnect in time".

I could see depth data, RGB Camera and Motion data and play around with it for a while. So it seemed to work at first. 

Now I can't turn the depth data on anymore, the GUI says "No frames Received!". The error log shows:
```
 03/07 00:59:25,402 ERROR [0x70000c38b000] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 03/07 00:59:25,402 ERROR [0x70000c61a000] (sensor.cpp:517) acquire_power failed: failed to set power state
 03/07 00:59:25,402 ERROR [0x70000c61a000] (error-handling.cpp:78) Error during polling error handler: failed to set 
```power state

A hardware reset in the viewer doesn't resolve the issue, nor does a power cycle of the iMac or the sensor. I actually physically measured the power draw of the device to between 500 and 700mW.

Changing the preset with data turned off, resulted in a time out error.

The firmware log then shows messages interrupted by:
```
RealSense error calling rs2_get_fw_log(dev:0x7fef90808f90, fw_log_msg:0x7fef90a0cda0):
    failed to set power state

Waiting for RealSense device to connect...
RealSense device was connected...
```

Is there anything else that I can test?
Is it a firmware issue? Should I reflash it?

Best Regards
Johannes

PS:

rs-enumerate-devices
```
 03/07 00:33:49,752 ERROR [0x700004aec000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
 03/07 00:33:49,752 ERROR [0x700004d7b000] (sensor.cpp:517) acquire_power failed: failed to set power state
 03/07 00:33:49,752 ERROR [0x700004d7b000] (error-handling.cpp:78) Error during polling error handler: failed to set power state
Device info: 
    Name                          : Intel RealSense L515
    Serial Number                 : f1150534
    Firmware Version              : 01.05.08.01
    Recommended Firmware Version  : 01.05.08.01
    Physical Port                 : 20-8-9
    Debug Op Code                 : 15
    Product Id                    : 0B64
    Camera Locked                 : YES
    Usb Type Descriptor           : 3.2
    Product Line                  : L500
    Asic Serial Number            : 0003ba0ad033
    Firmware Update Id            : 0003ba0ad033
 
Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Confidence   1024x768 @ 30Hz   RAW8
    Confidence   640x480 @ 30Hz   RAW8
    Confidence   320x240 @ 30Hz   RAW8
    Infrared   1024x768 @ 30Hz   Y8
    Infrared   640x480 @ 30Hz   Y8
    Infrared   320x240 @ 30Hz   Y8
    Depth   1024x768 @ 30Hz   Z16
    Depth   640x480 @ 30Hz   Z16
    Depth   320x240 @ 30Hz   Z16
 
Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color   1920x1080 @ 30Hz   RGB8
    Color   1920x1080 @ 30Hz   Y16
    Color   1920x1080 @ 30Hz   BGRA8
    Color   1920x1080 @ 30Hz   RGBA8
    Color   1920x1080 @ 30Hz   BGR8
    Color   1920x1080 @ 30Hz   YUYV
    Color   1920x1080 @ 15Hz   RGB8
    Color   1920x1080 @ 15Hz   Y16
    Color   1920x1080 @ 15Hz   BGRA8
    Color   1920x1080 @ 15Hz   RGBA8
    Color   1920x1080 @ 15Hz   BGR8
    Color   1920x1080 @ 15Hz   YUYV
    Color   1920x1080 @ 6Hz   RGB8
    Color   1920x1080 @ 6Hz   Y16
    Color   1920x1080 @ 6Hz   BGRA8
    Color   1920x1080 @ 6Hz   RGBA8
    Color   1920x1080 @ 6Hz   BGR8
    Color   1920x1080 @ 6Hz   YUYV
    Color   1280x720 @ 60Hz   RGB8
    Color   1280x720 @ 60Hz   Y16
    Color   1280x720 @ 60Hz   BGRA8
    Color   1280x720 @ 60Hz   RGBA8
    Color   1280x720 @ 60Hz   BGR8
    Color   1280x720 @ 60Hz   YUYV
    Color   1280x720 @ 30Hz   RGB8
    Color   1280x720 @ 30Hz   Y16
    Color   1280x720 @ 30Hz   BGRA8
    Color   1280x720 @ 30Hz   RGBA8
    Color   1280x720 @ 30Hz   BGR8
    Color   1280x720 @ 30Hz   YUYV
    Color   1280x720 @ 15Hz   RGB8
    Color   1280x720 @ 15Hz   Y16
    Color   1280x720 @ 15Hz   BGRA8
    Color   1280x720 @ 15Hz   RGBA8
    Color   1280x720 @ 15Hz   BGR8
    Color   1280x720 @ 15Hz   YUYV
    Color   1280x720 @ 6Hz   RGB8
    Color   1280x720 @ 6Hz   Y16
    Color   1280x720 @ 6Hz   BGRA8
    Color   1280x720 @ 6Hz   RGBA8
    Color   1280x720 @ 6Hz   BGR8
    Color   1280x720 @ 6Hz   YUYV
    Color   960x540 @ 60Hz   RGB8
    Color   960x540 @ 60Hz   Y16
    Color   960x540 @ 60Hz   BGRA8
    Color   960x540 @ 60Hz   RGBA8
    Color   960x540 @ 60Hz   BGR8
    Color   960x540 @ 60Hz   YUYV
    Color   960x540 @ 30Hz   RGB8
    Color   960x540 @ 30Hz   Y16
    Color   960x540 @ 30Hz   BGRA8
    Color   960x540 @ 30Hz   RGBA8
    Color   960x540 @ 30Hz   BGR8
    Color   960x540 @ 30Hz   YUYV
    Color   960x540 @ 15Hz   RGB8
    Color   960x540 @ 15Hz   Y16
    Color   960x540 @ 15Hz   BGRA8
    Color   960x540 @ 15Hz   RGBA8
    Color   960x540 @ 15Hz   BGR8
    Color   960x540 @ 15Hz   YUYV
    Color   960x540 @ 6Hz   RGB8
    Color   960x540 @ 6Hz   Y16
    Color   960x540 @ 6Hz   BGRA8
    Color   960x540 @ 6Hz   RGBA8
    Color   960x540 @ 6Hz   BGR8
    Color   960x540 @ 6Hz   YUYV
    Color   640x480 @ 60Hz   RGB8
    Color   640x480 @ 60Hz   Y16
    Color   640x480 @ 60Hz   BGRA8
    Color   640x480 @ 60Hz   RGBA8
    Color   640x480 @ 60Hz   BGR8
    Color   640x480 @ 60Hz   YUYV
    Color   640x480 @ 30Hz   RGB8
    Color   640x480 @ 30Hz   Y16
    Color   640x480 @ 30Hz   BGRA8
    Color   640x480 @ 30Hz   RGBA8
    Color   640x480 @ 30Hz   BGR8
    Color   640x480 @ 30Hz   YUYV
    Color   640x480 @ 15Hz   RGB8
    Color   640x480 @ 15Hz   Y16
    Color   640x480 @ 15Hz   BGRA8
    Color   640x480 @ 15Hz   RGBA8
    Color   640x480 @ 15Hz   BGR8
    Color   640x480 @ 15Hz   YUYV
    Color   640x480 @ 6Hz   RGB8
    Color   640x480 @ 6Hz   Y16
    Color   640x480 @ 6Hz   BGRA8
    Color   640x480 @ 6Hz   RGBA8
    Color   640x480 @ 6Hz   BGR8
    Color   640x480 @ 6Hz   YUYV
    Color   640x360 @ 60Hz   RGB8
    Color   640x360 @ 60Hz   Y16
    Color   640x360 @ 60Hz   BGRA8
    Color   640x360 @ 60Hz   RGBA8
    Color   640x360 @ 60Hz   BGR8
    Color   640x360 @ 60Hz   YUYV
    Color   640x360 @ 30Hz   RGB8
    Color   640x360 @ 30Hz   Y16
    Color   640x360 @ 30Hz   BGRA8
    Color   640x360 @ 30Hz   RGBA8
    Color   640x360 @ 30Hz   BGR8
    Color   640x360 @ 30Hz   YUYV
    Color   640x360 @ 15Hz   RGB8
    Color   640x360 @ 15Hz   Y16
    Color   640x360 @ 15Hz   BGRA8
    Color   640x360 @ 15Hz   RGBA8
    Color   640x360 @ 15Hz   BGR8
    Color   640x360 @ 15Hz   YUYV
    Color   640x360 @ 6Hz   RGB8
    Color   640x360 @ 6Hz   Y16
    Color   640x360 @ 6Hz   BGRA8
    Color   640x360 @ 6Hz   RGBA8
    Color   640x360 @ 6Hz   BGR8
    Color   640x360 @ 6Hz   YUYV
 
Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel N/A @ 400Hz   MOTION_XYZ32F
    Accel N/A @ 200Hz   MOTION_XYZ32F
    Accel N/A @ 100Hz   MOTION_XYZ32F
    Gyro N/A @ 400Hz   MOTION_XYZ32F
    Gyro N/A @ 200Hz   MOTION_XYZ32F
    Gyro N/A @ 100Hz   MOTION_XYZ32F
```

MartyG-RealSense, 2021-07-03:
Hi @jkneer  The firmware driver and hardware of your camera looks okay.  As a starting point in investigating your case, could you try unplugging the USB cable from the end that is inserted in the side of the camera, turning the connector around the opposite way and re-inserting it please (the cable has 2-way insertion at the camera end of it).  Does flipping the direction of the connector make a positive difference?

jkneer, 2021-07-03:
I could successfully flash (at least that's what the tool says) the 1.5.5.0 firmware without falling into recovery mode:
```
rs-fw-update -l

connected devices:
1) Name: Intel RealSense L515, serial number: f1150534, update serial number: 0003ba0ad033, firmware version: 01.05.05.00, USB type: 3.2 
```
But I still observe the same behaviour as described above.

Also I tried turning the USB-C connector - without effect.

I reactivated an ubuntu machine and installed librealsense: I could install the newest firmware through the viewer.
Yet I still observe the same behaviour (in the viewer: Fatal error occured (16) ).

MartyG-RealSense, 2021-07-03:
Can you confirm that you do not receive depth data even if you set a very low resolution of **320x240** depth and do not activate the RGB or Motion Module?

[image]

jkneer, 2021-07-03:
[image]

1. 
No depth data, RGB data and IMU data work.

rs-capture also fails:
```
./examples/capture/rs-capture 
 03/07 17:12:47,570 ERROR [0x110b3adc0] (handle-libusb.h:95) failed to claim usb interface: 7, error: RS2_USB_STATUS_ACCESS
 03/07 17:12:47,572 ERROR [0x110b3adc0] (handle-libusb.h:95) failed to claim usb interface: 7, error: RS2_USB_STATUS_ACCESS
 03/07 17:13:03,198 ERROR [0x110b3adc0] (handle-libusb.h:95) failed to claim usb interface: 7, error: RS2_USB_STATUS_ACCESS
 03/07 17:13:03,201 ERROR [0x110b3adc0] (handle-libusb.h:95) failed to claim usb interface: 7, error: RS2_USB_STATUS_ACCESS
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x7fbf7b762b50):
    Frame didn't arrive within 15000
```

2. It is an Intel iMac 2020, 3,8Ghz 8-Core Intel Core i7.

MartyG-RealSense, 2021-07-03:
I located a case where a couple of people reported a similar problem with using L515 on MacOS (similar errors, and being able to access RGB and Motion Module but not the depth stream).

https://github.com/IntelRealSense/librealsense/issues/7302

Some of the RealSense users involved in that case also experienced their camera reporting a serial number of 'ffffffffffffffff' in the Viewer, though you do not seem to be having that issue.  You can confirm by left-clicking on the **Info** option near the top of the Viewer's side-panel.

jkneer, 2021-07-03:
[image]

Yes, I had read that thread. The serialnumber reported corresponds to the one on the packaging.
What throws me off though, is the Firmware Update Id / Asic Serial number: are they supposed to be the same? Is it a coincidence that it reads *ba*d*?

MartyG-RealSense, 2021-07-03:
It is normal for the 'Asic Serial Number' and the 'Firmware Update Id' number of RealSense cameras to be the same in the Info section of the Viewer.  The ba*ad elements of the serial numbers would be coincidental and not significant.

jkneer, 2021-07-04:
At least nothing obvious, is there anything I should be looking out for?
The error of the sensor is the same on the iMac and on an Ubuntu machine, so for now I would guess, the problem lies with the sensor?

MartyG-RealSense, 2021-07-04:
If the Viewer is the first program that is launched after the computer has been booted then that should ensure that there is nothing else running that is claiming access to the camera when performing tests of it.

It looks as though the last time you performed a firmware update, you did it with the **rs-fw-update** tool.  Is that correct, please?  If so, let's use the alternative updater tool in the Viewer instead.

First, please download the **1.5.8.1** firmware bin file.

https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads

Then launch the RealSense Viewer and go to the **More** option near the top of the Viewer's options side-panel and select the **Update Firmware** option from the drop-down menu.

A file selector will appear.  Select the 1.5.8.1 firmware bin file from the location on your computer that you downloaded it to.  The camera should then begin updating its firmware to 1.5.8.1 in the Viewer.

If the update is able to complete successfully, are you then able to access the depth stream of your L515?

jkneer, 2021-07-04:
I did successfully flash firmware using rs-fw-update and using the viewer before.

Following your instructions, I get the following ( I rebooted right before doing so):

[image]
[image]
[image]

:(

MartyG-RealSense, 2021-07-04:
A RealSense user with a Windows computer has just reported the same L515 error in a new case at https://github.com/IntelRealSense/librealsense/issues/9333

jkneer, 2021-07-05:
Something caught my attention: The Apd Temperature displays as -9999, this should probably be a two-digit value
[image]

MartyG-RealSense, 2021-07-05:
When the Viewer is first launched, the **APD Temperature** is -9999 by default when the **L500 Depth Sensor** is disabled in the Viewer.  When it is enabled it remains at -9999 for a short time (about 5 seconds) and then should update to a proper value such as '24' the first time that a check of the temperature is performed.  It then returns to -9999 when L515 Depth Sensor is disabled, 

Does your APD temperature value update from -9999 if you wait 5 seconds or so, please?

jkneer, 2021-07-05:
No it never updates.

MartyG-RealSense, 2021-07-05:
Okay, the APD reading on my L515 looks like this after it updates.

[image]

I will include your APD temperature symptom in my report to Intel today.

jkneer, 2021-07-05:
Here is a log file on debuglevel that I recorded when I went through enabling and disabling the three data streams:
[librealsense.log](https://github.com/IntelRealSense/librealsense/files/6764026/librealsense.log)

MartyG-RealSense, 2021-07-05:
Thanks very much @jkneer for your detailed feedback.

jkneer, 2021-07-07:
I received an answer from the Intel support, and they confirmed that the device is faulty.

MartyG-RealSense, 2021-07-07:
Thanks for the update @jkneer

MartyG-RealSense, 2021-07-12:
Hi @jkneer  After further discussion with Intel, an official bug report relating to the **Fatal error occured (16)** error has been filed with Intel.  This case should be kept open whilst the bug report is active.  Thanks again for reporting the issue!

MartyG-RealSense, 2021-10-06:
Hi @jkneer  I discussed your case further with Intel.  If you still have the Fatal Error described in this case, they have requested that you return your camera to the retailer that you purchased it from, using the **Return Material Authorization (RMA)** product returns procedure of that particular retailer.

Also, could you please email me the serial number of your L515 camera.  You can find the serial number by clicking the **Info** option near the top of the RealSense Viewer's options side-panel.  The email address to send the serial number to is:

martyx.grover@intel.com

jkneer, 2021-10-06:
Hi,

the device was RMAed back in July, the new device is working.

MartyG-RealSense, 2021-10-06:
Hi @jkneer  Thanks very much for the update!  I'm pleased to hear that your new L515 is working.

This case shold be kept open as there is still an Intel bug report associated with it, but you do not need to take any further action with this case yourself.  Thanks again!

MartyG-RealSense, 2022-02-13:
Case closed due to successful receiving of a replacement camera in July 2021.

closed 2022-02-13 by MartyG-RealSense