realsenseai/librealsense #9330: Dispatcher error when building librealsense from source MacOS

JoeHRIsaac, 2021-07-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |  MacOS Big Sur | 
| Platform                                 | PC|
| SDK Version                            | 2.48.0                     | 
| Language                            |  python                       | 

### Issue Description
When I'm trying to build the librealsense from source, the make files get generated with no problem (with CMake gui) but after running make -j4 I get this error:

Consolidate compiler generated dependencies of target fw
Consolidate compiler generated dependencies of target realsense-file
[  1%] Built target fw
Consolidate compiler generated dependencies of target pybackend2
[  4%] Built target libcurl
[  4%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/rsusb-backend/rsusb-backend.cpp.o
[  4%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/__/__/src/hid/hid-device.cpp.o
[ 10%] Built target realsense-file
Consolidate compiler generated dependencies of target realsense2
[ 10%] Building CXX object CMakeFiles/realsense2.dir/src/hid/hid-device.cpp.o
[ 10%] Building CXX object CMakeFiles/realsense2.dir/src/rsusb-backend/rsusb-backend.cpp.o
[ 11%] Linking CXX shared library pybackend2.cpython-38-darwin.so
Undefined symbols for architecture x86_64:
  "dispatcher::stop()", referenced from:
      librealsense::notifications_processor::~notifications_processor() in types.cpp.o
      librealsense::notifications_processor::set_callback(std::__1::shared_ptr<rs2_notifications_callback>) in types.cpp.o
      librealsense::platform::rs_uvc_device::~rs_uvc_device() in uvc-device.cpp.o
      librealsense::platform::uvc_streamer::flush() in uvc-streamer.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::stop() in uvc-streamer.cpp.o
      librealsense::platform::rs_hid_device::~rs_hid_device() in hid-device.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::stop() in hid-device.cpp.o
      ...
  "dispatcher::start()", referenced from:
      librealsense::notifications_processor::set_callback(std::__1::shared_ptr<rs2_notifications_callback>) in types.cpp.o
      librealsense::platform::rs_uvc_device::rs_uvc_device(std::__1::shared_ptr<librealsense::platform::usb_device> const&, librealsense::platform::uvc_device_info const&, unsigned char) in uvc-device.cpp.o
      librealsense::platform::uvc_streamer::uvc_streamer(librealsense::platform::uvc_streamer_context) in uvc-streamer.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::start() in uvc-streamer.cpp.o
      librealsense::platform::rs_hid_device::rs_hid_device(std::__1::shared_ptr<librealsense::platform::usb_device>) in hid-device.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::start() in hid-device.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::start() in rsusb-backend-linux.cpp.o
      ...
  "dispatcher::dispatcher(unsigned int, std::__1::function<void (std::__1::function<void (dispatcher::cancellable_timer const&)>)>)", referenced from:
      librealsense::notifications_processor::notifications_processor() in types.cpp.o
      librealsense::platform::rs_uvc_device::rs_uvc_device(std::__1::shared_ptr<librealsense::platform::usb_device> const&, librealsense::platform::uvc_device_info const&, unsigned char) in uvc-device.cpp.o
      librealsense::platform::uvc_streamer::uvc_streamer(librealsense::platform::uvc_streamer_context) in uvc-streamer.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::active_object(std::__1::function<void (dispatcher::cancellable_timer)>) in uvc-streamer.cpp.o
      librealsense::platform::rs_hid_device::rs_hid_device(std::__1::shared_ptr<librealsense::platform::usb_device>) in hid-device.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::active_object(std::__1::function<void (dispatcher::cancellable_timer)>) in hid-device.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::active_object(std::__1::function<void (dispatcher::cancellable_timer)>) in rsusb-backend-linux.cpp.o
      ...
  "dispatcher::~dispatcher()", referenced from:
      librealsense::notifications_processor::~notifications_processor() in types.cpp.o
      librealsense::platform::rs_uvc_device::rs_uvc_device(std::__1::shared_ptr<librealsense::platform::usb_device> const&, librealsense::platform::uvc_device_info const&, unsigned char) in uvc-device.cpp.o
      librealsense::platform::rs_uvc_device::~rs_uvc_device() in uvc-device.cpp.o
      librealsense::platform::uvc_streamer::uvc_streamer(librealsense::platform::uvc_streamer_context) in uvc-streamer.cpp.o
      librealsense::platform::uvc_streamer::~uvc_streamer() in uvc-streamer.cpp.o
      active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::~active_object() in uvc-streamer.cpp.o
      librealsense::platform::rs_hid_device::rs_hid_device(std::__1::shared_ptr<librealsense::platform::usb_device>) in hid-device.cpp.o
      ...
  "_hid_close", referenced from:
      librealsense::platform::rs_hid_device::~rs_hid_device() in hid-device.cpp.o
  "_hid_enumerate", referenced from:
      librealsense::platform::rs_hid_device::rs_hid_device(std::__1::shared_ptr<librealsense::platform::usb_device>) in hid-device.cpp.o
  "_hid_get_feature_report", referenced from:
      librealsense::platform::rs_hid_device::hidapi_PowerDevice(unsigned char) in hid-device.cpp.o
  "_hid_open", referenced from:
      librealsense::platform::rs_hid_device::rs_hid_device(std::__1::shared_ptr<librealsense::platform::usb_device>) in hid-device.cpp.o
  "_hid_read", referenced from:
      librealsense::platform::rs_hid_device::handle_interrupt() in hid-device.cpp.o
  "_hid_send_feature_report", referenced from:
      librealsense::platform::rs_hid_device::hidapi_PowerDevice(unsigned char) in hid-device.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wrappers/python/pybackend2.2.48.0.cpython-38-darwin.so] Error 1
make[1]: *** [wrappers/python/CMakeFiles/pybackend2.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 12%] Linking CXX shared library librealsense2.dylib
[ 46%] Built target realsense2

What is this error? What should I install to fix it?

MartyG-RealSense, 2021-07-03:
Hi @JoeHRIsaac  Is your Mac a recent model that uses the **Apple Silicon M1** Arm-based processor instead of the Intel processor that earlier Macs used, please?

If it is an M1 Mac, the installation process may be more complicated than that of an Intel-based Mac.  The best references for an M1 installation are the links below.

https://github.com/IntelRealSense/librealsense/issues/8250
https://github.com/IntelRealSense/librealsense/issues/8090

Otherwise, if you are building for an older Intel-based Mac then I recommend looking at Intel's official MacOS installation page if you have not done so already.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

JoeHRIsaac, 2021-07-03:
@MartyG-RealSense 

It's an Intel Mac. And yes I was following those steps and exact lines with the addition of python bindings since I want pyrealsense2

MartyG-RealSense, 2021-07-03:
It does look as though the build process is exiting when it reaches the Python wrapper.

```
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [wrappers/python/pybackend2.2.48.0.cpython-38-darwin.so] Error 1
make[1]: *** [wrappers/python/CMakeFiles/pybackend2.dir/all] Error 2
```

A RealSense user shared in the link below the process that worked for them personally when building the SDK and Python wrapper with CMake.

https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-643304216

JoeHRIsaac, 2021-07-05:
Ok, the problem seem to be cleared when I use -DPYTHON_EXECUTABLE and specify my python explicitly. The following worked: Thanks a lot for the support.

git clone https://github.com/IntelRealSense/librealsense
brew install cmake libusb pkg-config
cd librealsense
mkdir build
cd build
cmake .. DBUILD_PYTHON_BINDINGS=true -DPYTHON_EXECUTABLE=/Users/joeisaac/opt/anaconda3/bin/python
make -j4
make install

closed 2021-07-05 by JoeHRIsaac

MartyG-RealSense, 2021-07-05:
Great news @JoeHRIsaac - thanks very much for the update and for sharing the solution that worked for you with the RealSense community!