realsenseai/librealsense #9333: Fatal Error (16)

Elle-Yen, 2021-07-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |01.05.08.01| 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  None                                        | 
| Platform                                 | PC|
| SDK Version                            | 2.0  | 
| Language                            |  C                      | 
| Segment			|  Intel RealSense Viewer v2.47.0                  | 

### Issue Description
The L500 Depth Sensor sometimes doesn't work.
It gave me "Fatal error occurred (16)". I found its relative kind "ld_alram_hard_error" in codes "librealsense/src/l500/l500-private.h". What's this error? Is there any explaination about the kinds of errors listed in "l500_fw_error_report"?

MartyG-RealSense, 2021-07-04:
Hi @Elle-Yen   There is a currently open L515 case with this error at https://github.com/IntelRealSense/librealsense/issues/9329 that has been encountered on Mac and Ubuntu computers by the RealSense user in that case.  I see that you have encountered it on Windows.

Elle-Yen, 2021-07-04:
Thanks for your reply. I have read #9329, but still don't understand. What's the meaning of "ld_alarm_hard_error"? Is there any detailed explaination?

MartyG-RealSense, 2021-07-04:
There is not an available definition for **ld_alarm_hard_error**.  I will highlight the issue to Intel tomorrow but I do not currently have information to offer on this particular subject.

Elle-Yen, 2021-07-04:
Thank you very much for your help.

Elle-Yen, 2021-07-05:
I'm sorry for disturbing. May I have some more detailed explaination about the types in "l500_fw_error_report" such as "ld_alarm_hard_error"? Thanks a lot.

MartyG-RealSense, 2021-07-05:
It is no trouble at all.  I have reported this issue to Intel today and will update you as soon as I have feedback to provide.  Thanks very much for your patience!

MartyG-RealSense, 2021-07-12:
Hi @Elle-Yen  Thanks very much again for your patience.  After discussion with Intel, it was decided to file an official bug report with Intel in regard to **ld_alarm_hard_error** so that it can be investigated by the RealSense team.

I would recommend following https://github.com/IntelRealSense/librealsense/issues/9329 - the case that the bug report is linked to - in order to receive updates about this error and closing your own case here.

closed 2021-07-24 by MartyG-RealSense