realsenseai/librealsense #9335: DIM-RealSense Activation Tool Issue

vedigovind, 2021-07-05:
Camera Model                       : L515
 Firmware Version                  : 01.05.04.01 
 Operating System & Version : Win 10
 
I tried to create a DIM license file using the product key, but I encountered an error, please help me resolve this issue.
[image]

RealSenseSupport, 2021-07-24:
Hi,

We apologize for the difficulty in activating your copy of Intel® RealSense™ Dimensional Weight Software.

In order to resolve your issue, we need:
1) The serial number of the Intel® RealSense™ LiDAR Camera L515 you used to activate the software
2)The email address you used to register the software.
3) The product key (in your activation email) of the software.

Thanks

vedigovind, 2021-07-27:
Thanks for your reply, your source is already taking care of it.

closed 2021-09-14 by RealSenseSupport