realsenseai/librealsense #9337: It's strange while using other cable

berylyellow, 2021-07-05:
Required Info                        

| Camera Model                       |  D455                       | 
| Firmware Version                   |  05.12.07.150           | 
| Operating System & Version  |  Ubuntu 18.04         | 
| Kernel Version (Linux Only)    |  4.18.0-17-generic   | 
| Platform                                 |  PC   &   rk3399       |
| SDK Version                           | 2.47.0                       | 
| Language                               | Python   &    c++    | 

I found it very strange.  

Firstly, I just installed pyrealsense2 by pip and successed. And fetch frames without error by using official 1-meter cable. But when I use my 5-meter cable, it always returns: Frame didn't arrive within 15000 or Frame didn't arrive within 5000 without showing a frame. I have almost tried all means you mentioned in other issues but failed. Maybe it does not support to use other cable?

Then, I try to build c++ examples from source by :
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6  -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true

After building c++ examples successfully, I run realsense-viewer in my pc, it shows that: 
"The camera was detected by the OS as connected to a USB 2.1 port".  But my cable has USB 3.0 port.  

After running realsense-viewer, I try to run opencv_viewer_example.py by python3, it returns : 
File "opencv_viewer_example.py", line 18, in <module>
    pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected

Then I try to reboot my pc, and run opencv_viewer_example.py by python3 directly, it shows frames unsmoothly for several seconds but then an error occurrs :
File "opencv_viewer_example.py", line 45, in <module>
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000

Then I try to run python code again after running realsense-viewer, it always returns "No device connected". And failed to run rs-capture by c++ with error log “Frame didn't arrive within 15000”.

I also try to use other platform rk3399 to test, but strangely,when I run realsense-viewer, it shows:
"The camera was detected by the OS as connected to a USB 3.2 port".

Why? My cable has 3.0 port and so does my pc. 
In conclusion, 
it is different when running realsense-viewer on rk3399 and pc, my cable with USB 3.0 port can not be recognized.
And device cannot be recognizied by using Python code after running realsense-viewer.
When running python code alone, "Frame didn't arrive within 5000" or "Frame didn't arrive within 15000".

Looking forward to your reply, thanks!

MartyG-RealSense, 2021-07-05:
Hi @berylyellow  Because RealSense cameras can transmit a high volume of data along the USB cable, having a high quality cable for lengths greater than 2 meters is important, otherwise the camera connection can degrade or disconnect.  An example of 'industrial grade' cables suitable for RealSense are the ones supplied by a company called Newnex.

https://www.newnex.com/realsense-3d-camera-connectivity.php
https://www.youtube.com/watch?v=GLQgR1jT04M

roelofvandijk, 2021-07-05:
Hi @berylyellow, these devices can be quite difficult to extend to longer distances properly. I have tried USB 3.0 to USB 3.0 extension cables that work, notably active ones, but my current favourite choice is a 3m USB 3.0 to USB-C cable from Anker ([Amazon DE link](https://amzn.to/2UqJoYh)) for ~18 Euro. Make sure you are buying cables that support at least USB 3.1 Gen 1 (Superspeed) with a rated throughput of 5 Gbit/s, and consider active USB extension cables such as [this one](https://amzn.to/2TCY6Lt) if you need longer distances.

MartyG-RealSense, 2021-07-05:
Thanks very much @roelofvandijk for your Anker recomendation!

Another cable brand that has been recommended in the past by RealSense 400 Series customers is **OCP**.

https://www.ocp.com/product/usb-3-0-3-1-cable-type-c-to-a-gen1-5g-black-intel-d435-realsense-camera-62-00214/

berylyellow, 2021-07-06:
Thanks very much @MartyG-RealSense @roelofvandijk  for your reply and advice.

Finally I solved it by using a 5m USB 3.0 to USB 3.0 HUB with independent power supply.

But there is still a question, I cannot run python code after running realsense-viewer. Why does this happen？

MartyG-RealSense, 2021-07-06:
Excellent news @berylyellow that you solved your cable problem.

Do you have the RealSense Viewer open and a stream enabled in it when you run your Python code?   If so, the Viewer does not need to be open in order to run programs that you have created yourself, and if a particular stream is enabled in the Viewer then that stream becomes inaccessible to other prograns until disabled in the Viewer.  This is because the first program to enable a stream 'claims' exclusive access to it until that stream is disabled, releasing the claim on it.

MartyG-RealSense, 2021-07-06:
You could try inserting a **hardware_reset()** instruction into your Python script to reset the camera when the script launches.

 https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

berylyellow, 2021-07-08:
Hi @MartyG-RealSense  thanks for your help.
I have tried inserting hardware_reset() as you mentioned, but unfortunately it didn't work and repoted that：
 File "opencv_viewer_example.py", line 30, in <module>
    pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected

It seems it didn't connect to the camera.

Anyway, thanks for your kindly help. Since the main cable problem has been solved, I will close this issue.

closed 2021-07-08 by berylyellow

MartyG-RealSense, 2021-07-08:
Thanks very much @berylyellow for the update!

Marjaneh-T, 2022-10-06:
@berylyellow I have the same issue for D455 and need a 5m cable.
Can you please mention the HUB you bought with more specifications?

Thanks
Marjaneh